ros/ros_comm #543: roscpp: Empty connection header topic

tonybaltovski, 2015-01-03:
I am trying to use a driver that reads the name of the incoming topic from the connection header.  When I publish data from a C++ node, it is always empty but when I use `rostopic pub` or python it works just fine.  Is there a way to ensure the connection header is populated?  

OS: Ubuntu 14.04

ROS:indigo 1.11.9

roscpp: 1.11.9-0trusty-20141029-0127-+0000

roscpp_core: 0.5.4-0trusty-20140724-0053-+0000

The subscriber is in C++ and is [here](https://github.com/smd-ros-devel/lynxmotion_ssc32/blob/master/src/ssc32_driver.cpp#L391)

Here is the code in [python](https://gist.github.com/tonybaltovski/e1470fc1175b423ca5f4) and the [C++ node](https://gist.github.com/tonybaltovski/ac75f7e3273e7086a433)

gerkey, 2015-04-02:
Also, thanks much to @wjwwood for the easy-to-use test case.

wjwwood, 2015-04-03:
> Also, thanks much to @wjwwood for the easy-to-use test case.

Sure, no problem. Like my always said: "Do unto others".

closed 2015-04-03 by dirk-thomas

tonybaltovski, 2015-06-04:
Thanks guys!

PR #597 (2015-04-03): Pass through the topic field of an incoming connection header, issue #543.
fixes #543