ros/ros_comm #546: rostest: add_rostest function should expose the underlying target

czalidis, 2015-01-13:
`add_rostest` function should expose the underlying target so we can add dependencies between targets explicitly. This could be useful when we want to add a dependency between a rostest and a target defined by a `catkin_download_test_data` call.

Currently, when running `catkin_make run_tests_my_pkg` catkin scans the dependencies of rostest targets in `my_pkg` but does not know that a file needs to be downloaded in order to run the test, so the test fails.

On the other hand, we can run `catkin_make tests` before we call the `run_tests_my_pkg` and the `catkin_download_test_data` executes properly because it has a dependency to `tests` target, so maybe this is the way to do it.

wjwwood, 2015-01-14:
Hi @czalidis, I agree with you that it should expose the target somehow. The only question is how.

The target name is created deterministically in this code block and stored as `_target_name`:

https://github.com/ros/ros_comm/blob/indigo-devel/tools/rostest/cmake/rostest-extras.cmake.em#L34-51

Ever since https://github.com/ros/ros_comm/pull/456 this target name has been based on the arguments to rostest as well as the file name and test type and package, etc... I believe this target name namespacing is a good thing to preserve.

One option would be to allow the user to specify a target name, something like this:

```cmake
add_rostest(... TARGET_NAME my_target_name ...)
```

But this would make the test names inconsistent and more likely to collide with another when being used with `catkin_make` (not a problem with `catkin_make_isolated`).

Therefore I would suggest just exposing the automatically generated name, which you could do with another free function like `get_rostest_name(output_var_name add_rostest_args)`. But if you did that you would need to redundantly pass the arguments to both this function and `add_rostest`.

So finally I would recommend adding an optional argument to `add_rostest` like `STORE_TARGET_NAME_IN` which then takes a variable name to store the target name in, so you would end up with something like this:

```cmake
add_rostest(path/to/foo.test ... STORE_TARGET_NAME_IN foo_target_name ...)
add_dependencies(${foo_target_name} ...)
```

I don't think we have time to dig in and create a pull request for this, complete with tests and documentation. However, if you wanted to try and do so, hopefully this description of the issue will help you get started.

dirk-thomas, 2015-01-14:
Imo the API should allow the user to optionally pass a custom target name.
While the default makes sense the API should not restrict the user.

But the last proposal from William (enabling to retrieve the target name) is a good idea, too.
But we should name the argument `TARGET_VARIABLE` which is more in line with other CMake arguments.

czalidis, 2015-01-15:
@wjwwood  the code section that you mention actually does not create any targets it self. There is a call to `catkin_run_tests_target()` which creates and handles the targets and defines dependencies between them. This function is part of the `catkin` package and can be found here: 

https://github.com/ros/catkin/blob/indigo-devel/cmake/test/tests.cmake#L92-144

So rostest does not know about the underlying target, catkin handles that part the same way that handles other `catkin_add_*test()` calls. 

Also, I noticed that the `catkin_run_tests_target()` has an optional argument `DEPENDENCIES`, so there is the ability to add dependencies between the underlying rostest target and any other (user defined or not) target exposing the same `DEPENDENCIES` argument to the `add_rostest()` function and let `catkin_run_tests_target()` to do the work.

Does this sound like a good solution?

As for adding the option to pass a custom target, we can do what `catkin_add_gtest()` does. Define a custom target, exactly as the user passed it, and also define a target for run tests using the `catkin_run_tests_target()`, adding a dependency between them.

https://github.com/ros/catkin/blob/indigo-devel/cmake/test/gtest.cmake#L25-59

wjwwood, 2015-01-15:
> the code section that you mention actually does not create any targets it self.

That's true, but it computes the target's name, which is all you need to add dependencies to the target.

> Also, I noticed that the catkin_run_tests_target() has an optional argument DEPENDENCIES, so there is the ability to add dependencies between the underlying rostest target and any other (user [...]

That's fine with me, but the only issue is that if people want/need the target name for something other than adding dependencies we're back to square one. That being said the `DEPENDENCIES` option being added to the `add_rostest` target should work just fine for this issue.

dirk-thomas, 2015-01-21:
@czalidis Can you please update the documentation in the following places (as referenced in the PR):
* http://docs.ros.org/indigo/api/catkin/html/howto/format2/rostest_configuration.html
* http://wiki.ros.org/rostest/Writing

czalidis, 2015-01-22:
Ok, I will update the documentation over the next few days.

Any chance of backporting this to hydro?

dirk-thomas, 2015-01-22:
Great, thanks. Please post the links to the diff once you found the time for it.

Yes, since it's straight forward extension of API I will cherry pick the commit before the next hydro release.

czalidis, 2015-01-27:
I have updated the documentation in both places. 

Here is the diff for the catkin documentation: https://github.com/ros/catkin/pull/712/files

dirk-thomas, 2015-01-27:
Thanks. Both patches have been merged.

I will keep this open until it is backported to Hydro.

peci1, 2022-02-07:
I think this can be closed.

closed 2022-02-09 by wjwwood