ros/ros_comm #548: Control maximum logfile size

abarobopec, 2015-01-15:
Hello, 

I wrote a question on ros::answer some days ago about how to keep control of the logfile size. http://answers.ros.org/question/200611/is-it-possible-to-control-the-size-of-log-file/

I read carefully source code of roslaunch, and I understoud how the log file are managed. 

For me the problem, is that log file are generated by redirecting stdout and stderr of a subprocess in a file.

I propose to patch roslaunch to redirect stdout and stderr of the node subprocess in a python.logging.logger object that will be created with a RotatingFile handler. Paramters of this RotatingFile handlers are "maximum log file size" and "maximum log file count". These paramters will be two new attribute of the XML tag "node".

I will do a pull request very soon, thanks to review it and tell me what do you think about it.

Regards,

dirk-thomas, 2015-01-15:
I can see this feature being useful. You might also want to look at the pending PR #418. Since for this feature the flow to stdout / stderr needs to be controlled the same way it might make sense to build on-top of it.

abarobopec, 2015-01-16:
Ok, I will integrate the feature of PR #418 in my patch. It's quite easy, I just have to add a logging.handler to the logger if output=="both" ....

I Think I will be ready to commit it today.

closed 2025-05-31 by sloretz