ros/ros_comm #552: Add 'quiet' command line option

ehuang3, 2015-01-18:
While using roslaunch with `gdb -i=mi` the initial status output from roslaunch affects parsing of the machine interpreter interface. A quiet option to roslaunch would be very helpful.

Currently, I am using an ugly hack that replaces `sys.stdout` with my own stream that does no printing.

If someone more knowledge than I could point me at how to implement this the right way, I'd be happy to do so.

tfoote, 2015-01-18:
Can you clarify what you're trying to debug? Are you trying to debug a node or roslaunch itself? And give an example of how to reproduce the problem?

ehuang3, 2015-01-18:
I'm debugging a node using `emacs` builtin `gdb` functionality. To reproduce the problem, do `M-x gdb` and instead of executing `gdb -i=mi ...`, replace that with a roslaunch that has

    
    <arg name="debug" default="false" />
    <arg unless="$(arg debug)" name="launch_prefix" value="" />
    <arg     if="$(arg debug)" name="launch_prefix"
  	   value="gdb -i=mi -x $(find apc_moveit_config)/launch/gdb_settings.gdb --args" />

enabled.

closed 2025-05-31 by sloretz