ros/ros_comm #553: Improve documentation of rostest code coverage functionality

VGonPa, 2015-01-20:
`rostest.rosrun` seems to provide a code coverage tool that can be called from the `sysargs==--cov`.
However this functionality it is not documented and it is not clear how to use it.

See this [question](http://answers.ros.org/question/181892/rostest-code-coverage/?answer=200872#post-id-200872) in ROS Answers.

Some indications on the *docstring* of the function might help, although an example of use would be  much better.

dirk-thomas, 2015-01-20:
I won't have time to improve the documentation in the near future.

Please consider to contribute to the ROS wiki and/or propose a specific change here to get feedback before.

VGonPa, 2015-01-23:
I can do a PR to include this addition to the documentation of `rostest.rosrun` and later, perhaps update the wiki. But to be sure how it works I been trying to get some coverage of my tests without success.

I've found a bug (#558) to which I've already submitted a PR (#559).
However it is still not clear to me why it is possible to call rostest with `--text`, but not with `--cov`(you need to put `--cov` inside the `sysargs` parameter of `rostest.rosrun`.

felixdivo, 2018-02-13:
Has this simply died?

defied, 2020-01-31:
Seems like it. Where does the coverage output go?

njoye, 2020-03-31:
This would be really useful information as well as functionality, pleeaase make it work.... I have searched for multiple days now and I can't seem to get to run any kind of code coverage for our ros python projects...

mariabohorquez, 2020-08-07:
There is a package that can help: https://github.com/mikeferguson/code_coverage

njoye, 2020-08-08:
> There is a package that can help: https://github.com/mikeferguson/code_coverage

I have used that package to succesfully solve my problem and create code coverage for my ROS project. The `code_coverage` package now also works for python projects!

Even though the package exists: I don't think that a third-party package that solves an issue exonerates non-existent or bad documentation for the original package. So this Issue seems to still be valid.

RafaelRey, 2023-03-03:
Hi, I'm using the mentioned code_coverage package for ROS CPP Unit Tests and I'm getting the coverage report successfully, howewer, I would like to know if it is possible to get code coverage also for Rostests implemented using gtest

closed 2025-05-31 by sloretz