ros/ros_comm #557: rospy publisher slows down with multiple clients

kshehata, 2015-01-22:
As first reported in [ROS Answers] (http://answers.ros.org/question/201524/publisher-send-rate-drops-with-multiple-clients-in-rospy/), if a rospy publisher tries publishing at a high rate (e.g. 100 Hz) it works well with 1 client. As soon as a few more clients are added, the publisher slows to a crawl.

I've made the [ros_py_tests](https://github.com/kshehata/ros_py_tests) package to demonstrate the effect. It creates a publisher that sends out Headers at 100 Hz, and then subscribers that check the timestamps for sync and sequence numbers for any dropped messages. If you run a publisher with 5 subscribers (use the launch files, it's much easier), you'll see the message rate drop to nothing. For example:

1. Launch the py-py-tests.launch to get a publisher and 5 subscribers
2. rostopic hz / test

Rate will drop to something around 5-10 Hz

A few interesting points:
* Seems to work fine on VM's
* Seems to consistently fail on Ubuntu 12.04
* Testing with a mix of C++ and Python, seems like it's a problem with rospy's publisher. A C++ publisher doesn't have the problem, and C++ subscribers don't make a difference.
* The time stamps seem to stay in sync, which seems to suggest that the publisher is getting blocked on sending

This is causing a lot of headaches because the applanix_driver is based on rospy and broadcasts at 100 Hz. As soon as multiple clients are listening, the message rate from the driver drops to 30-40 Hz.

dirk-thomas, 2015-01-22:
Thank you for the detailed report and the example package. I ran the py-py-test.launch and `rostopic hz` reports constant 100 Hz.

Can you please provide more information:
* which version of ROS you are using as well as
* the version numbers of the some of the involved core packages (e.g. `rospy`)

kshehata, 2015-01-22:
Definitely ROS Hydro on Ubuntu 12.04. I'll get specific version numbers ASAP. (Unfortunately doesn't happen on my machine, since I'm running Mac / VM).

Tiv6, 2015-01-22:
Here's the versions of some of the involved packages:

rospy: 1.10.2-0precise-20140304-0128-+0000
ros-comm: 1.10.2-0precise-20140304-0227-+0000

dirk-thomas, 2015-01-22:
@Tiv6 The current version of `rospy` in Hydro is: 1.10.12-0

Please update you system and check if the problem is still the same with up-to-date versions of ROS packages.

Tiv6, 2015-01-23:
Issue was resolved by updating.

kshehata, 2015-01-23:
Looks like somewhere between rospy 1.10.2-0 and 1.10.11-0 was an update that fixed this. Thanks for the quick response @dirk-thomas!

closed 2015-01-23 by kshehata