ros/ros_comm #561: rosconsole: NAMED logger

OkkeHendriks, 2015-01-29:
I am not completely sure if the following is correct or incorrect behavior. 

I have a node which has multiple instances of a certain class. I want to be able to individually set the logger level of each instance. In order to achieve this I figured i could use ROS_INFO_NAMED(name1,...) ROS_INFO_NAMED(name2, ...), etc..

However the name I see on the screen or in rqt logger levels is name1. This is not what I expected.

I wrote a simple testcase:

https://gist.github.com/OkkeHendriks/4f4adfeafb71875ed75b

Is this a bug?

wjwwood, 2015-01-29:
Here's the implementation:

https://github.com/ros/ros_comm/blob/ce75bf8d9c19f8337d2196d3d19f9ba8745e0583/tools/rosconsole/include/rosconsole/macros_generated.h#L100

It's clear from the implementation, that this macro will take the first parameter as a literal. Since all of the `NAME`'ed macros are implemented this way, I don't think it is a bug with the implementation of just this macro. However, I could see why you might want something that interprets the `name` parameter of the macro as a `std::string` variable.

OkkeHendriks, 2015-01-30:
Ah yes, sorry, this happens not only with ROS_INFO_NAMED, but with all NAME'ed macro's.

dirk-thomas, 2015-02-02:
The current implementation allows you to either:

* pass a plain string:

    ROS_INFO_NAMED("foo", "msg");

* or pass a variable name (which should be of type char* /  std::string:

    std::string varname = "foo";
    ROS_INFO_NAMED(varname, "msg");

Both will lead to a rosconsole logger named `ros.NODENAME.foo`.

@OkkeHendriks I don't understand what exactly the problem is you are having. Can you plain describe it in more detail, e.g.  with a self contained code snippet.

wjwwood, 2015-02-02:
I agree with @dirk-thomas, I was reading the implementation incorrectly before.

OkkeHendriks, 2015-02-03:
I will try to elaborate further by first giving a simple example in which I see this behavior and then explaining the situation in which I encountered this problem.

#### Example ####

The following code demonstrates the problem. An easy to clone package can be found [here](https://github.com/OkkeHendriks/ros_named_logger/tree/master).
This example uses ROS_INFO_NAMED but the same happens with DEBUG, WARN, ERROR and FATAL.

```c++
#include <ros/ros.h> 
class TestClass
{
  public:
	TestClass(const std::string name)
	{
		ROS_INFO_NAMED(name, "TestClass: This logger should be named %s?", name.c_str());
	}
 
	~TestClass(){};
	
};
 
int main(int argc, char *argv[])
{
	ros::init(argc, argv, "Node_Name");
	ros::NodeHandle n;
	
	std::string name1 = "Name1";
	std::string name2 = "Name2";
	
	ROS_INFO_NAMED(name1, "Main: This logger should be named %s?", name1.c_str());
	ROS_INFO_NAMED(name2, "Main: This logger should be named %s?", name2.c_str());
 
	TestClass test1(name1);
	TestClass test2(name2);
}
```

Expected output:
> ros.ros_named_logger_test.Name1[ INFO]: Main: This logger should be named Name1?
> ros.ros_named_logger_test.Name2[ INFO]: Main: This logger should be named Name2?
> ros.ros_named_logger_test.Name1[ INFO]: TestClass: This logger should be named Name1?
> ros.ros_named_logger_test.Name2[ INFO]: TestClass: This logger should be named Name2?

Obtained output:
> ros.ros_named_logger_test.Name1[ INFO]: Main: This logger should be named Name1?
> ros.ros_named_logger_test.Name2[ INFO]: Main: This logger should be named Name2?
> ros.ros_named_logger_test.Name1[ INFO]: TestClass: This logger should be named Name1?
> ros.ros_named_logger_test.**Name1**[ INFO]: TestClass: This logger should be named Name2?

#### Use case ####
I am making a 'Server Multiple Client' wrapper class which contains a single SimpleActionServer and multiple ComplexClients. Each ComplexClient is another class which wraps a SimpleActionClient. 
All the ComplexClients have there own `std::string name`, I would like to be able see which ComplexClient printed a certain message, and be able to set their logger levels separately, by using the NAMED loggers.

dirk-thomas, 2015-02-03:
I can reproduce the problem. A simplified version of the issue:

    #include <ros/ros.h>
    void func(const std::string& other) {
        ROSCONSOLE_DEFINE_LOCATION(true, ::ros::console::levels::Info, std::string(ROSCONSOLE_NAME_PREFIX) + "." + other);
        ROSCONSOLE_PRINT_AT_LOCATION("func: this logger should be named %s?", other.c_str());
    }
    int main(int argc, char *argv[]) {
        func("beingused");
        func("notbeingused");
    }

dirk-thomas, 2015-02-03:
The problem here is how `ROSCONSOLE_DEFINE_LOCATION` registers new loggers. It does that by managing static variables based on the `name` argument. But since the value of the name argument is not known a compile time the created logger stays the same - even when later being used with different values. All this infrastructure is necessary to enable configuring the loggers later (e.g. via a GUI), toggling individual loggers, changing their levels, etc.

I don't expect this to be fixable in the current design. This should also be documented in the wiki:
[ ] http://wiki.ros.org/rosconsole/CodeAPI

OkkeHendriks, 2015-02-04:
Basically it should initialize a new logger when the name changes? 
Could [`ROSCONSOLE_DEFINE_LOCATION`](http://docs.ros.org/hydro/api/rosconsole/html/console_8h.html#a0f904d70b0c9d729b0e6682b5656f38a) keep track of and compare the actual name variable, instead of only checking the `initialized_` boolean?

Initializing a new logger whenever the name changes.

dirk-thomas, 2015-02-04:
Yes, it would be possible to keep a static list of loggers keyed by their name. But that would require quite some changes as well as testing. It is not likely that I will have the time to develop a feature like that. If you would like to work on a PR I would be happy to review it.

dirk-thomas, 2015-02-04:
You could also try to use the API behind the macros directly to make it work just for your use case.

dirk-thomas, 2015-03-20:
I have a added a note to the wiki page to make users aware of this limitation: http://wiki.ros.org/action/diff/rosconsole/CodeAPI?action=diff&rev1=3&rev2=4

dirk-thomas, 2015-03-20:
Since this is currently the "intended" behavior I would close this ticket. If you feel strongly about it please consider providing a PR to support "dynamic" named loggers. But as mentioned before I assume this to be pretty difficult to integrate.

closed 2015-03-20 by dirk-thomas