ros/ros_comm #571: ROS_DEBUG_DTHROTTLE

galou, 2015-02-19:
Sometimes I want to show a message that a node is waiting for a message on a specific topic. For that, I use ROS_DEBUG_THROTTLE.

However, ROS_DEBUG_THROTTLE shows a message on first call. I would like to have a Delayed (hence DTHROTTLE, name to be discussed), that would first show the message after the specified rate elapsed.

I'm OK to write a PR after the name has been discussed.

dirk-thomas, 2015-02-19:
I am not sure I understand the expected behavior. Can you please clarify. If the macro is only being called once but not again later you want the message to be sent after a delay?

galou, 2015-02-19:
The behavior would be the same as ROS_DEBUG_THROTTLE with the exception of the time of the first message:
ROS_DEBUG_THROTTLE(1, ...) sends at t = 0, 1, 2, 3, ...
ROS_DEBUG_DTHROTTLE(1, ...) would send at t = 1, 2, 3, ...
where t = 0 is the time where the macro is called for the first time with this message.

Does it make sense?

dirk-thomas, 2015-02-19:
Ok, I think I get the idea.

The current `THROTTLE` macros are described with:

    Throttled output will print a message at most once per "period".

The implementation is pretty straight forward. For each message the current time is compared to the last printed message and if the duration exceeds the interval the message is actually printed and the last timestamp updated.

How would you described the new macro and how would you imagine the implementation to work?

galou, 2015-02-19:
The implementation is straight forward, I think,

in include/ros/console.h, copy ROS_LOG_THROTTLE to ROS_LOG_DTHROTTLE and replace
```c++
    static double last_hit = 0.0; \
    ::ros::Time now = ::ros::Time::now(); \
```
with
```c++
    ::ros::Time now = ::ros::Time::now(); \
    static double last_hit = now.toSec(); \
```

For the description, I would write something like:
```
DThrottled output will print a message at most once per "period" and at first after "period".
```

galou, 2015-02-19:
I didn't really replied one question: "If the macro is only being called once but not again later you want the message to be sent after a delay?"
No the user would need to repeatedly call the macro, either the macro outputs something right at the moment when it's called or nothing.
Typically the macro would be called inside a loop where no message need to be sent before a first "period" elapsed, in the case that the problem solves by itself in this time interval.

dirk-thomas, 2015-02-19:
Ok, as you said if in the loop it calls it a couple of times but less then the duration it doesn't print anything. Does that imply that if then much later a single call happens it will be immediately printed?

galou, 2015-02-21:
I guess this would make sense to print immediately if the macro is called again after a long period of time, because this would not be the first call, for behavior consistency.

galou, 2015-03-10:
@dirk-thomas Any further comment?

dirk-thomas, 2015-03-10:
That sounds like a reasonable behavior. It would be great if you could provide a PR for that.

dirk-thomas, 2015-03-10:
Regarding the name I think `DTHROTTLE` is not very clear. Instead it could be named `DELAYED_THROTTLE`?

galou, 2015-03-11:
You're right about the clarity of the name. But with your proposed name, this would give a very long name for the longest macro `ROS_DEBUG_STREAM_DELAYED_THROTTLE_NAMED`. Would it be OK?

dirk-thomas, 2015-03-11:
I don't have a better proposal at the moment. So, ok.

dirk-thomas, 2015-03-20:
Can you please update the wiki page describing the macros to mention the newly added ones? You should annotate this with (added in Indigo as of `rosconsole` version 1.11.11.

galou, 2015-03-23:
Done!

closed 2015-03-23 by galou