ros/ros_comm #573: roscpp service calls leak memory

grieneis, 2015-02-25:
We recently upgraded some code from 12.04/Fuerte to 14.04/Indigo.  After running our code over the weekend (~60 hours), we saw that the memory usage of a node that acted as a service server increased from ~50 mb to 10 gb.  Further investigation showed that it was due to the (non-persistent) service calls happening at 20 Hz.  I created a simplified test case with two nodes.  The first node advertises a service with the std_srvs/Empty message and the second node calls it at 20 Hz.  The simplified test case demonstrates the same rate of memory leaking as our more complicated code.

I ran both test nodes in valgrind, and they both leak an equal amount of memory.

I am not sure how to attach my files here, but I added my test nodes and valgrind logs to my answer to this ROS answers question:  http://answers.ros.org/question/199753/ros-serviceclient-memory-leak/.

Tests were run on Ubuntu 14.04 with ROS Indigo with valgrind version valgrind-3.10.0.SVN.

wjwwood, 2015-02-25:
I created a package with your test in it:

https://github.com/wjwwood/test_ros_comm_issue_573

I was unable to get the same output from valgrind on my Mac, but I haven't tried on Linux. What version of `ros_comm` are you using?

grieneis, 2015-02-25:
Thanks for making a package.  I'm using ros_comm version 1.11.10.

grieneis, 2015-02-25:
It looks like someone has already submitted a pull-request to fix this problem:  https://github.com/ros/ros_comm/pull/570

wjwwood, 2015-02-25:
@grieneis I was just arriving at the same conclusion, can we close this as a duplicate of that issue?

grieneis, 2015-02-25:
Yeah, I didn't see it when I made this issue.

wjwwood, 2015-02-25:
Duplicate of #570.

closed 2015-02-25 by wjwwood