ros/ros_comm #575: [roslaunch] Feature request: Tab completion for defined args

carlosjoserg, 2015-03-01:
Recently, I started to design and treat launch files in a similar as an APIs+library. That is, I create an _interface_ with args, and then I _implement_ the launch, to allow re-use of the same launch.

It would be nice if defined args (those with default values as `<args name="my_arg" default="my_value">`) pop up with tab completion, so one can see the interface from the command line in a similar way all comm does.

dirk-thomas, 2015-03-20:
Please consider to provide a pull request for this feature.

carlosjoserg, 2015-04-10:
Apologies, I didn't know about [this](https://github.com/ros/ros_comm/pull/230), [this](https://github.com/ros/ros_comm/pull/379) and [this](https://github.com/ros/ros/pull/19), looks that it does what I'm proposing. 

But it seems to be available only in `groovy`, is that right? If so, could you tell the reason of not being in `indigo`?

dirk-thomas, 2015-04-10:
Two of the PRs are from 2013. I am would assume that they are also in Hydro and Indigo. Can you point to anything why you assume they are not in Indigo?

carlosjoserg, 2015-04-10:
Sure, because I'm on `indigo` and I don't have that functionality.

See, the [`doc` attribute PR](https://github.com/ros/ros_comm/pull/379) is on the `indigo-devel` branch, and the`--ros-args` parameter works for me, but completion does not. [This PR](https://github.com/ros/ros/pull/19) was merged in the `groovy-devel` branch.

You can re-confirm by searching the `rosbash` script in the [`groovy-devel`](https://github.com/ros/ros/blob/groovy-devel/tools/rosbash/rosbash#L490) and [`indigo-devel`](https://github.com/ros/ros/blob/indigo-devel/tools/rosbash/rosbash) branches by the term `--ros-args`, that will take you to where the completion was implemented in the former whereas you will get no results in the latter.

dirk-thomas, 2015-04-10:
It seems like that changes have been merged to groovy-devel while the Hydro release was in progress. And the patch hasn't been merged to Hydro and therefore is not available in Indigo. It looks that also applies to the dependent PR on `ros_comm`.

I will look into it in the next days and determine which parts needs to be merged into newer branches.

dirk-thomas, 2015-04-13:
I merged two commits from *groovy-devel*:

* to *hydro-devel*:
  * https://github.com/ros/ros/commit/4e8943f8362c97876b9e56ea39bb98df3b9e809c
  * https://github.com/ros/ros/commit/d966b227227b81d715c4b41e84b0aa4b1150e0e9
* to *indigo-devel*:
  * https://github.com/ros/ros/commit/86f1ae7e0d8a042e2c110f75f7ab8823485f9110
  * https://github.com/ros/ros/commit/16e39d686174da9138a1c8983e8da4a702999154

Completion works for me now with the Indigo branch. Thanks for pointing it out.

closed 2015-04-13 by dirk-thomas