ros/ros_comm #579: Rostest with non-standard GTest location and no catkin_add_gtest does not compile

kyle-f, 2015-03-11:
`add_rostest_gtest` currently relies upon Catkin's gtest.cmake file to define `GTEST_FOUND`. If that variable is found, rostest will create the new test target. However, what is perhaps unintended is that `add_rostest_gtest` also relies on Catkin's `catkin_add_gtest` function to include any non-standard gtest include and lib directories (note that Catkin supports gtest installs found via `find_package`, so this is a likely scenario).

If one were to create a package with only `add_rostest_gtest` tests and _no_ `catkin_add_gtest` calls, along with using find_package to get a GTest installed in a non-standard location, the rostest compilation will fail due to an inability to find GTest headers and libs.

dirk-thomas, 2015-03-12:
The problem with `gtest` is that it is *not* sufficient to `find_package` it. It always involves to also build the gtest libraries which catkin is doing for you. Using precompiled gtest libraries is highly discouraged (see the gtest docs for an explanation of the reason).

Therefore I don't think the provided patch is sufficient. Instead the `add_rostest_gtest` function should perform parts of `catkin_add_gtest` (specifically https://github.com/ros/catkin/blob/7c2682f8921eddcb149213edf601bcf4ccf30cfd/cmake/test/gtest.cmake#L26-L51) and then register the launch file as a rostest.

kyle-f, 2015-03-12:
@dirk-thomas Agreed, good call. Rather than duplicating that code, how would you feel about a patch to Catkin to draw those lines into a separate function callable by rostest? Then I can modify this issue's patch to make use of it.

dirk-thomas, 2015-03-12:
Sure, that makes most sense. I don't know when I will find the time to do it but if you could provide a PR for catkin to split out the highlighted part of the function into something like `catkin_create_gtest` (?) that would certainly speed things up.

kyle-f, 2015-03-12:
Yeah, no problem! I'll take care of it.

dirk-thomas, 2015-04-02:
@kyle-f Could you please give the PR ros/catkin#726 a try and confirm that it satisfies your use case.

dirk-thomas, 2015-04-13:
I will close this ticket for now since it seems to be fixed by the referenced PR. If not please feel free to comment with more information.

closed 2015-04-13 by dirk-thomas