ros/ros_comm #585: roswtf crashes when trying to output warning about local hostname resolution with IPv6 address

neurovertex, 2015-03-23:
System : Archlinux [Linux 3.18.6-1-ARCH #1 SMP PREEMPT Sat Feb 7 x86_64 GNU/Linux]
Python version : 2.7.9
Trace:
```
Loaded plugin tf.tfwtf
No package or stack in context
rosinstall -- 
You are missing core ROS Python modules: 
Traceback (most recent call last):
  File "/opt/ros/indigo/bin/roswtf", line 35, in <module>
    roswtf.roswtf_main()
  File "/opt/ros/indigo/lib/python2.7/site-packages/roswtf/__init__.py", line 89, in roswtf_main
    _roswtf_main()
  File "/opt/ros/indigo/lib/python2.7/site-packages/roswtf/__init__.py", line 175, in _roswtf_main
    roswtf.network.wtf_check(ctx)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roswtf/network.py", line 109, in wtf_check
    error_rule(r, r[0](ctx), ctx)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roswtf/rules.py", line 102, in error_rule
    _check_rule(rule, ret, ctx, ctx.errors, WtfError)
  File "/opt/ros/indigo/lib/python2.7/site-packages/roswtf/rules.py", line 73, in _check_rule
    ctx_list.append(level(f_msg%d + ret%d, f_msg, ret))
ValueError: unsupported format character 'w' (0x77) at index 66
```

By adding a few print()'s, I found out that the error came from the second parameter passed to error_rule in network.py:109 : "Local hostname [Bubblegum] resolves to [fe80::2216:d8ff:fe42:95ff%wlp2s0], which does not appear to be a local IP address ['127.0.0.1', '*redacted*', '*redacted*'].". The string contains a literal %, which crashes upon formatting in _check_rule.

Easily avoidable by patching network.py :
```
52c52
<     global_ips = [ ip for ip in resolved_ips if not ip.startswith('127.') and not ip == '::1']
---
>     global_ips = [ ip.replace('%', '%%') for ip in resolved_ips if not ip.startswith('127.') and not ip == '::1']
```

j-rivero, 2015-04-02:
I assume that this problem comes from the fack of using zone index in your IP address, right?

neurovertex, 2015-04-03:
It does seem so, since wlp2s0 is my wireless card (named by systemd)

dirk-thomas, 2015-04-13:
Can you please try i a build from source using the latest state of the indigo-devel branch fixes the problem for you? If not, can you please provide more information about the exact values which lead to the stacktrace (you would need to add some debug output)?

dirk-thomas, 2015-04-14:
Closing this ticket since the referenced PR should fix the problem.

closed 2015-04-14 by dirk-thomas