ros/ros_comm #587: odometry messages to C++ node silently dropped

sambishop, 2015-03-25:
I hope I am not wasting anyone's time, but I have a very simple and reproducible issue where I am able to subscribe to odometry messages using Hydro but not using Indigo.  I posted this to ROS Answers first (http://answers.ros.org/question/205741/odometry-messages-to-c-node-silently-dropped/) along with details, including my simple test case and steps to reproduce the problem, but no one was able to help me.

It still seems likely that I am doing something wrong, but it also seems odd that it works with Hydro but doesn't using Indigo--without any warning or error messages.

sambishop, 2015-03-25:
One more thing.  I have reproduced this behavior (works with strings but doesn't work with an odometry message) on two computers running Indigo.  I did that using the trivial ROS package I have posted to GitHub.  You can get that using with `svn co https://github.com/sambishop/aav-ros-car/trunk/aav_car_ws/src/aav_control2`.  (It's a Subversion command because Subversion will let you checkout a sub-directory of a project, but Git doesn't.  You're welcome to checkout the whole project though--it's pretty small.)

dirk-thomas, 2015-03-25:
Please see answer on answers.ros.org.

closed 2015-03-25 by dirk-thomas