ros/ros_comm #589: Adding description field to a rosparam

TheDash, 2015-03-27:
I've commonly run into the issue where I'd like to know what a rosparam does, and what are the effects of changing it. 

ideally, I'd like to see the command

rosparam info /some/ros/param

print out something like this:

Loaded from: { foo.launch | foo_node | foo_user}
Description: changing this param does blah blah blah
Nodes using this parameter: [node1, node2, node3]
History: [value1 at time N, value2 at time N+1] 

Is this possible, or in the pipeline?

dirk-thomas, 2015-03-27:
None of this is currently possible. A parameter has:
* no knowledge of its origin
* no description attached
* is not aware which nodes queried it or
* no history of the values

Nothing is planned in this direction. Are you considering to work on patches to work toward this goal? It sounds like a lot of things to tackle.

TheDash, 2015-03-27:
Yes, I have been considering these features and wanted to make sure it hadn't been done before. it sounds like a lot, but would bring a lot of improvement to the system in terms of documentation, auto-configuration, system diagnostic, and so forth. 

This ticket can be marked as solved, just wanted to see where we were at. Thanks @dirk-thomas  :-)

closed 2015-03-27 by dirk-thomas