ros/ros_comm #591: Alternative version of ros::param::param with return value

xperroni, 2015-03-30:
The template function `ros::param::param` sets a variable either to the value of a parameter, or if not available, a predefined default value. In cases where the variable is passed straight to a constructor or function call and then disregarded, more streamlined code could be written if the result value could be retrieved as a function return value, rather than through an output variable.

For example, the code below:

    std::string path;
    std::string format;
    double fps;

    param("~path", path, "video.mpg");
    param("~format", format, "MPEG");
    param("~fps", fps, 30.0);

    VideoRecorder recorder(path, format, fps);

Could be more succinctly written as:

    VideoRecorder recorder(
        param<std::string>("~path", "video.mpg"),
        param<std::string>("~format", "MPEG"),
        param<double>("~fps", 30.0)
    );

If an alternative definition of `ros::param::param` such as this was available:

    template<typename T>
    T param(const std::string& param_name, const T& default_val)
    {
      T param_val;
      param(param_name, param_val, default_val);
      return value;
    }

dirk-thomas, 2015-03-30:
Looks reasonable. Could you please provide a pull request adding this feature? It would be great if you would also include a unit test for it to ensure its working / continues to work in the future.

The existing function is implemented in two locations:
* clients/roscpp/include/ros/param.h
* clients/roscpp/include/ros/node_handle.h

xperroni, 2015-03-31:
Done. I wasn't aware of the param method in NodeHandle or the unit test suite, thank you for the heads-up.

xperroni, 2015-04-07:
Pull request sent and merged.

closed 2015-04-07 by xperroni

tfoote, 2015-04-07:
For reference it was #592