ros/ros_comm #593: [feature request] Add additional conditional arguments ifempty/unlessempty

piyushk, 2015-04-02:
I've often found that I want a different behavior if an argument is empty or not. For instance, 
```
<arg name="tf_prefix" default="" />
<arg unlessempty="$(arg tf_prefix)" name="rgb_frame_id" default="$(arg tf_prefix)/$(arg camera)_rgb_optical_frame" />
<arg ifempty="$(arg tf_prefix)" name="rgb_frame_id" default="$(arg camera)_rgb_optical_frame" />
```

1. Is this a good idea? Is there a way I can achieve this with the current roslaunch xml syntax?
2. If I submit a PR with this feature, is it too late to be merged into Indigo?

gavanderhoorn, 2015-04-07:
Is this similar to #540?

piyushk, 2015-04-07:
@gavanderhoorn Looks like it. I'll follow up on that issue report.

closed 2015-04-07 by piyushk