ros/ros_comm #602: rostopic could warn users once that it is using sim time

guihomework, 2015-04-09:
Just a suggestion:

It would be nice to have a one time WARN/INFO message when calling rostopic hz telling if it is using simulated time. On rostopic echo this is quite obvious as the timestamps are not walltime (small values for seconds) but on rate aspects there is no such indication, one can easily be confused and the rate then seems perfect although the real rate is slower.

Here under is the context in which we found it could have been useful :
We have usually have simulation running at a realtime factor of 1.0, so we kind of forgot about simulation time and walltime differences and everything works fine when using a ROS only environment.

However, we also use another transport framework (RSB) in our environment that is not aware of simulation time. A few days ago, a new piece of code was slow and we (among others) used *rostopic hz* and the equivalent function of RSB to debug which part was slow (not publishing correctly)

Unfortunately it took a while to understand the new process was using more CPU than expected, making the realtime factor 0.3 on the simulation side (which we did not see as there was no movement involved) or overall slowdown, and hence we got the walltime and simtime to be streched by the same factor regarding rates. Both topic monitoring tools indicated different rates of publication and we did not realize this could come from the sim time, we thought our second transport framework had issues.

we found that after adding our own rate calculation in our loop, first with ros::time (which again failed to display anything wrong due to awareness of simtime) and then with non-ros timing functions that finally uncovered the problem.

An indication in rostopic hz at the very beginning might help to recall the rate we see is according to simulated time.

spaghetti-, 2015-08-24:
I'll take a look into it but if I'm not wrong we've to check if the node publishing the topic has `/use_sim_time` set. @wjwwood is there another approach that could work?

wjwwood, 2015-09-01:
@spaghetti- Yes, that is the most reliable way that I know of to see if simulation time is being used. I agree that a one time warning would be useful.

You could conceivably have `rostopic hz` show a simulation rate and a real time rate, but there might be issues implementing that given the possibility of time travel, e.g. when a bag file loops.

dirk-thomas, 2016-03-11:
This has been implemented in #672.

closed 2016-03-11 by dirk-thomas