ros/ros_comm #604: rosbag_storage fails to build on OS X yosemite

fferri, 2015-04-19:
Used to build fine on another machine with OS X 10.10.2.
Note that 122 ROS packages before this just build fine.

Configuration: OS X 10.10.3, ROS Indigo

[rosbag_storage] ==> '/Users/federico/indigo_desktop_ws/build/rosbag_storage/build_env.sh /usr/bin/make cmake_check_build_system' in '/Users/federico/indigo_desktop_ws/build/rosbag_storage'
[rosbag_storage] <== '/Users/federico/indigo_desktop_ws/build/rosbag_storage/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0'
[rosbag_storage] ==> '/Users/federico/indigo_desktop_ws/build/rosbag_storage/build_env.sh /usr/bin/make -j4 -l4' in '/Users/federico/indigo_desktop_ws/build/rosbag_storage'
[  9%] Building CXX object CMakeFiles/rosbag_storage.dir/src/bag_player.cpp.o
[ 18%] Building CXX object CMakeFiles/rosbag_storage.dir/src/query.cpp.o
[ 27%] Building CXX object CMakeFiles/rosbag_storage.dir/src/stream.cpp.o
[ 36%] Building CXX object CMakeFiles/rosbag_storage.dir/src/view.cpp.o
[ 45%] Building CXX object CMakeFiles/rosbag_storage.dir/src/uncompressed_stream.cpp.o
[ 54%] Building CXX object CMakeFiles/rosbag_storage.dir/src/message_instance.cpp.o
In file included from /Users/federico/indigo_desktop_ws/src/ros_comm/rosbag_storage/src/message_instance.cpp:28:
In file included from /Users/federico/indigo_desktop_ws/src/ros_comm/rosbag_storage/include/rosbag/message_instance.h:38:
In file included from /Users/federico/indigo_desktop_ws/src/roscpp_core/roscpp_traits/include/ros/message_traits.h:33:
In file included from /Users/federico/indigo_desktop_ws/src/roscpp_core/rostime/include/ros/time.h:58:
In file included from /usr/local/include/boost/math/special_functions/round.hpp:14:
In file included from /usr/local/include/boost/math/policies/error_handling.hpp:31:
In file included from /usr/local/include/boost/format.hpp:34:
/usr/local/include/boost/format/format_fwd.hpp:24:69: error: no type named 'allocator' in namespace 'std'
        class Tr = BOOST_IO_STD char_traits<Ch>, class Alloc = std::allocator<Ch> >
                                                               ~~~~~^
/usr/local/include/boost/format/format_fwd.hpp:24:78: error: expected ',' or '>' in template-parameter-list
        class Tr = BOOST_IO_STD char_traits<Ch>, class Alloc = std::allocator<Ch> >
                                                                             ^
/usr/local/include/boost/format/format_fwd.hpp:24:83: error: expected unqualified-id
        class Tr = BOOST_IO_STD char_traits<Ch>, class Alloc = std::allocator<Ch> >
                                                                                  ^
/usr/local/include/boost/format/format_fwd.hpp:27:13: error: unknown type name 'basic_format'
    typedef basic_format<char >     format;
            ^
/usr/local/include/boost/format/format_fwd.hpp:27:25: error: expected unqualified-id
    typedef basic_format<char >     format;
                        ^
/usr/local/include/boost/format/format_fwd.hpp:30:13: error: unknown type name 'basic_format'
    typedef basic_format<wchar_t >  wformat;
            ^
/usr/local/include/boost/format/format_fwd.hpp:30:25: error: expected unqualified-id
    typedef basic_format<wchar_t >  wformat;
                        ^
In file included from /Users/federico/indigo_desktop_ws/src/ros_comm/rosbag_storage/src/message_instance.cpp:28:
In file included from /Users/federico/indigo_desktop_ws/src/ros_comm/rosbag_storage/include/rosbag/message_instance.h:38:
In file included from /Users/federico/indigo_desktop_ws/src/roscpp_core/roscpp_traits/include/ros/message_traits.h:33:
In file included from /Users/federico/indigo_desktop_ws/src/roscpp_core/rostime/include/ros/time.h:58:
In file included from /usr/local/include/boost/math/special_functions/round.hpp:14:
In file included from /usr/local/include/boost/math/policies/error_handling.hpp:31:
In file included from /usr/local/include/boost/format.hpp:35:
/usr/local/include/boost/format/internals_fwd.hpp:32:5: error: unknown type name 'basic_format'
    basic_format<Ch, Tr, Alloc>&  
    ^
/usr/local/include/boost/format/internals_fwd.hpp:32:17: error: expected unqualified-id
    basic_format<Ch, Tr, Alloc>&  
                ^
/usr/local/include/boost/format/internals_fwd.hpp:37:5: error: unknown type name 'basic_format'
    basic_format<Ch, Tr, Alloc>&  
    ^
/usr/local/include/boost/format/internals_fwd.hpp:37:17: error: expected unqualified-id
    basic_format<Ch, Tr, Alloc>&  
                ^
/usr/local/include/boost/format/internals_fwd.hpp:48:10: warning: variable templates are a C++14 extension [-Wc++14-extensions]
    void distribute (basic_format<Ch,Tr, Alloc>& self, T x);
         ^
/usr/local/include/boost/format/internals_fwd.hpp:48:10: error: variable has incomplete type 'void'
/usr/local/include/boost/format/internals_fwd.hpp:48:22: error: use of undeclared identifier 'basic_format'
    void distribute (basic_format<Ch,Tr, Alloc>& self, T x);
                     ^
/usr/local/include/boost/format/internals_fwd.hpp:48:35: error: 'Ch' does not refer to a value
    void distribute (basic_format<Ch,Tr, Alloc>& self, T x);
                                  ^
/usr/local/include/boost/format/internals_fwd.hpp:47:20: note: declared here
    template<class Ch, class Tr, class Alloc, class T> 
                   ^
/usr/local/include/boost/format/internals_fwd.hpp:48:38: error: 'Tr' does not refer to a value
    void distribute (basic_format<Ch,Tr, Alloc>& self, T x);
                                     ^
/usr/local/include/boost/format/internals_fwd.hpp:47:30: note: declared here
    template<class Ch, class Tr, class Alloc, class T> 
                             ^
/usr/local/include/boost/format/internals_fwd.hpp:48:42: error: 'Alloc' does not refer to a value
    void distribute (basic_format<Ch,Tr, Alloc>& self, T x);
                                         ^
/usr/local/include/boost/format/internals_fwd.hpp:47:40: note: declared here
    template<class Ch, class Tr, class Alloc, class T> 
                                       ^
/usr/local/include/boost/format/internals_fwd.hpp:48:50: error: use of undeclared identifier 'self'
    void distribute (basic_format<Ch,Tr, Alloc>& self, T x);
                                                 ^
/usr/local/include/boost/format/internals_fwd.hpp:48:56: error: 'T' does not refer to a value
    void distribute (basic_format<Ch,Tr, Alloc>& self, T x);
                                                       ^
/usr/local/include/boost/format/internals_fwd.hpp:47:53: note: declared here
    template<class Ch, class Tr, class Alloc, class T> 
                                                    ^
/usr/local/include/boost/format/internals_fwd.hpp:51:5: error: unknown type name 'basic_format'
    basic_format<Ch, Tr, Alloc>& 
    ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
1 warning and 20 errors generated.
make[2]: *** [CMakeFiles/rosbag_storage.dir/src/message_instance.cpp.o] Error 1
make[1]: *** [CMakeFiles/rosbag_storage.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
[rosbag_storage] <== '/Users/federico/indigo_desktop_ws/build/rosbag_storage/build_env.sh /usr/bin/make -j4 -l4' failed with return code '2'

fferri, 2015-04-19:
I checked on the other machine (OS X 10.10.2), which previously built ROS Indigo fine.
Now it fails with this very same error.

I guess this is related to Xcode 6.3 / Xcode Command Line Tools 6.3. Now both machines have the toolchain provided by Xcode 6.3. I don't know how to downgrade to 6.2 or 6.1.

Here I found a different issue, but possibly related: http://stackoverflow.com/questions/29529455/missing-c-header-debug-after-updating-osx-command-line-tools-6-3

fferri, 2015-04-19:
I confirm that downgrading to Xcode 6.2 and Xcode Command Line Tools 6.2 solves this issue :-)

wjwwood, 2015-04-20:
Yeah, this seems to be a an issue with CLT, I believe Apple is aware of it and plans to fix it soon. For now either downgrade or you can create the missing file as described here:

http://stackoverflow.com/questions/29529455/missing-c-header-debug-after-updating-osx-command-line-tools-6-3/29576048#29576048

closed 2015-04-20 by wjwwood