ros/ros_comm #605: roscpp fails to build on OS X yosemite

fferri, 2015-04-19:
Used to build fine -some week ago- on another machine with OS X 10.10.2 (same compiler).
Note that 122 ROS packages before this just build fine.

Configuration:
OS: Mac OS X 10.10.3
Compiler: Apple LLVM version 6.1.0 (clang-602.0.49) (based on LLVM 3.6.0svn) x86_64-apple-darwin14.3.0
ROS: Indigo

[roscpp] ==> '/Users/federico/indigo_desktop_ws/build/roscpp/build_env.sh /usr/local/bin/cmake /Users/federico/indigo_desktop_ws/src/ros_comm/roscpp -DCATKIN_DEVEL_PREFIX=/Users/federico/indigo_desktop_ws/devel -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/include/python2.7' in '/Users/federico/indigo_desktop_ws/build/roscpp'
-- The C compiler identification is AppleClang 6.1.0.6020049
-- The CXX compiler identification is AppleClang 6.1.0.6020049
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /Users/federico/indigo_desktop_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Found PythonInterp: /usr/local/bin/python (found version "2.7.9")
-- Using PYTHON_EXECUTABLE: /usr/local/bin/python
-- Using default Python package layout
-- Found PY_em: /usr/local/lib/python2.7/site-packages/em.pyc
-- Using empy: /usr/local/lib/python2.7/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /Users/federico/indigo_desktop_ws/build/roscpp/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/local/bin/nosetests-2.7
-- catkin 0.6.11
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.57.0
-- Found the following Boost libraries:
--   signals
--   filesystem
--   system
-- roscpp: 1 messages, 3 services
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE
-- Looking for include file ifaddrs.h
-- Looking for include file ifaddrs.h - found
-- Looking for trunc
-- Looking for trunc - found
-- Configuring done
CMake Warning (dev) at CMakeLists.txt:119 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "rosgraph_msgs_gencpp" of target "roscpp" does not
  exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:119 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "std_msgs_gencpp" of target "roscpp" does not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.
[... 164 lines cut ...]
/usr/local/include/boost/format/internals_fwd.hpp:47:40: note: declared here
    template<class Ch, class Tr, class Alloc, class T>
                                       ^
/usr/local/include/boost/format/internals_fwd.hpp:48:50: error: use of undeclared identifier 'self'
    void distribute (basic_format<Ch,Tr, Alloc>& self, T x);
                                                 ^
/usr/local/include/boost/format/internals_fwd.hpp:48:56: error: 'T' does not refer to a value
    void distribute (basic_format<Ch,Tr, Alloc>& self, T x);
                                                       ^
/usr/local/include/boost/format/internals_fwd.hpp:47:53: note: declared here
    template<class Ch, class Tr, class Alloc, class T>
                                                    ^
/usr/local/include/boost/format/internals_fwd.hpp:51:5: error: unknown type name 'basic_format'
    basic_format<Ch, Tr, Alloc>&
    ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
[ 86%] Building CXX object CMakeFiles/roscpp.dir/src/libros/init.cpp.o
[ 88%] Building CXX object CMakeFiles/roscpp.dir/src/libros/subscription.cpp.o
1 warning and 20 errors generated.
make[2]: *** [CMakeFiles/roscpp.dir/src/libros/rosout_appender.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/roscpp.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
[roscpp] <== '/Users/federico/indigo_desktop_ws/build/roscpp/build_env.sh /usr/bin/make -j4 -l4' failed with return code '2'

fferri, 2015-04-19:
I checked on the other machine (OS X 10.10.2), which previously built ROS Indigo fine.
Now it fails with this very same error.

I guess this is related to Xcode 6.3 / Xcode Command Line Tools 6.3. Now both machines have the toolchain provided by Xcode 6.3. I don't know how to downgrade to 6.2 or 6.1.

Here I found a different issue, but possibly related: http://stackoverflow.com/questions/29529455/missing-c-header-debug-after-updating-osx-command-line-tools-6-3

fferri, 2015-04-19:
I confirm that downgrading to Xcode 6.2 and Xcode Command Line Tools 6.2 solves this issue :-)

wjwwood, 2015-04-20:
I'm going to close this as a duplicate of https://github.com/ros/ros_comm/issues/604#issuecomment-94274712

closed 2015-04-20 by wjwwood