ros/ros_comm #607: rospy publisher error needs newline

dawonn, 2015-04-20:
When a publisher is provided a message with bad data, the following exception is thrown. It is missing a newline between the exception error message and the message contents. This was misleading because I thought it was specifically saying there was a problem in the 'header' but it was actually in another field. Bonus points if you can add information to the error message that indicates which field had the bad data in it. :)

Traceback (most recent call last):
  File "./main.py", line 66, in <module>
    talker()
  File "./main.py", line 52, in talker
    pub.publish(msg_scan)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/topics.py", line 838, in publish
    raise ROSSerializationException(str(e))
rospy.exceptions.ROSSerializationException: <class 'struct.error'>: 'required argument is not a float
' when writing 'header:
  seq: 1
  stamp:
    secs: 1429550592   
    nsecs: 870158910   
  frame_id: LaserScan

dirk-thomas, 2015-04-20:
I can't find the exception string from you snippet. Can you please provide a simple reproducible example where this happens?

dawonn, 2015-04-20:
Thanks for the quick reply. :)

#!/usr/bin/env python
    # BSD License 
    import rospy
    from sensor_msgs.msg import LaserScan

    import csv
    import sys
    import IPython

    def talker():
        # Prepare a publisher
        pub = rospy.Publisher('scan', LaserScan, queue_size=10)
        rospy.init_node('mtu_laserlog', anonymous=True)
        rate = rospy.Rate(40) # hz
       
        while not rospy.is_shutdown():
                
            msg_scan = LaserScan()
            
            #msg_scan.header.seq   = 0;
            msg_scan.header.stamp = rospy.Time.now()
            msg_scan.header.frame_id = "LaserScan"
            
            msg_scan.angle_min = "this isn't right..."
            
            pub.publish(msg_scan)
            rate.sleep()
            
            #Ipython.embed()
            

    if __name__ == '__main__':
        try:
            talker()
        except rospy.ROSInterruptException:
            pass

dawonn, 2015-04-20:
Oh, and for what it's worth, the initial problem that I had was that I had set msg_scan.ranges = a list of strings rather than a list of floats. 

I fixed it with this one-liner:
            ranges[:] = [float(x)          for x in ranges]

dirk-thomas, 2015-04-20:
The actual exception is coming from https://github.com/ros/genpy/blob/b7a1c8fed043e2aef2f21b50fa69e0b35779f2be/src/genpy/generator.py#L682-L683

I do see your case that when the value is multiple lines it is more difficult to read. While the string block is quoted it can easily be missed. But in many other cases the value is only a single string where the additional newline would make it more difficult to read.

Therefore I lean towards not adding a newline in this exception string since it would only improve some cases but make others worse.

dirk-thomas, 2016-03-11:
Based on the argument of the previous comment I will close this for now. If you think it still needs fixing please consider providing a pull request for `genpy`.

closed 2016-03-11 by dirk-thomas