ros/ros_comm #608: ROS service interface for Rosbag

muhrix, 2015-04-21:
Hi all,

About a year ago I modified the `rosbag` package to act as a ROS service provider for local/remote logging control. That way I could start/stop/start-over the recording of messages being published via simple ROS service calls.

I believe this is a welcome feature to `rosbag`, and it is related to issue #428.

I have noticed there were changes to the `rosbag` source code since then. I had to make significant changes to the code in order to handle the queue of incoming messages and the fact that, once I stopped recording, the node would not shut down (service provider) and hence cause memory leaks when a new recording started.

Before I go into greater detail, I would like to ask what would be best in preparation for a release:

1. Release a separate package, with a different name
2. Release within `rosbag` an extra node which implements the service functionality
3. Release within `rosbag` and extend the current node with an optional parameter to start rosbag as a service

I will be tidying up the code and bringing it up to date with the latest changes, then make it publicly available soon-ish, but I would like to discuss the best approach so that a package with this functionality can be released.

@tfoote @dirk-thomas and everybody else, could you please share your points of view?

Many thanks!

tfoote, 2015-04-21:
I think that providing a separate node makes sense. For initial development it makes sense to use a separate package. But assuming it can be done without adding dependencies or conflicting names, promoting it into the rosbag package seams a reasonable goal.

muhrix, 2015-04-27:
I'll initially work on a separate package then. Thanks!

tfoote, 2015-04-28:
After thinking about this a bit, it might be even more powerful/flexible to design this to work as a nodelet. It could be launched in the same process to take advantage of the optimized intraprocess communications and possibly avoid network communications of the data entirely.

muhrix, 2015-04-30:
Do you mean attaching to existing publishers?
Wouldn't this assume publishers are implemented as nodelets too? (or am I missing something?)

In any case, both node and nodelet versions could be provided.

tfoote, 2015-04-30:
Yes it would only work if the publisher is a nodelet. But if implemented as a nodelet, it can be run as a standalone nodelet to provide the equivalent of a node without needing a parallel version.

DLu, 2015-11-25:
+1 
I would also like the functionality described in the original issue. Nodelet vs new node doesn't matter.

dirk-thomas, 2015-12-09:
Recently a rosbag nodelet has been developed which offers an action-based interface: https://github.com/osrf/nodelet_rosbag

Does this cover the requested feature and this ticket can be closed?

DLu, 2015-12-17:
Is there any documentation for this the nodelet_rosbag?

nourani, 2016-01-12:
An additional feature is delayed recording, where internally the node maintains a buffer (say 30sec) and only starts recording to disk once the trigger message has arrived. 

This is really useful when dealing with a large amount of data, can't afford logging permanently, but want to be able to trigger the log when observing something.

dirk-thomas, 2016-03-11:
@DLu There is already a ticket about the lack of documentation (osrf/nodelet_rosbag#2). Please feel free to comment on it.

@nourani Please fill a separate ticket about a `--delay` option for `rosbag record`. If you would like to have that for the action-based interface please fill it at https://github.com/osrf/nodelet_rosbag Please also consider to provide a pull request for the enhancement.

Since a separate already provides the requested functionality I will close this ticket. Please feel free to comment here or on the repository containing the new functionality.

closed 2016-03-11 by dirk-thomas