ros/ros_comm #609: Why ``rosnode`` does not allow ``l`` as ``list`` like ``rosservice`` and ``rostopic``?

wkentaro, 2015-04-22:
https://github.com/ros/ros_comm/blob/indigo-devel/tools/rosnode/src/rosnode/__init__.py#L790

`rostopic` and `rosservice` allows `l` like::
* https://github.com/ros/ros_comm/blob/indigo-devel/tools/rosservice/src/rosservice/__init__.py#L735
* https://github.com/ros/ros_comm/blob/indigo-devel/tools/rostopic/src/rostopic/__init__.py#L1770

Are there some reasons `rosnode` does not allow `l` as `list`?

dirk-thomas, 2015-04-22:
This is actually a bug in `rosservice` and `rostopic`. The following check:

    if command in 'list':

is wrong since when you call `rostopic i` it will actually invoke `list` instead of `info`.

Therefore the right way to address this will be to fix the comparion in `rosservice` and `rostopic` to compare the whole command string.

closed 2015-04-22 by dirk-thomas

wkentaro, 2015-04-23:
I see. thanks!

commit 7de7e249 (2015-04-22): fix command parsing for rosservice and rostopic to not except parts of 'list' (fix #609)