ros/ros_comm #612: roslaunch unknown host remote node

fevb, 2015-04-30:
Hi,

when using roslaunch to launch remote nodes one usually comes across a 'host not in SSH known_hosts file' error.

A partial solution is hinted here: http://answers.ros.org/question/41446/a-is-not-in-your-ssh-known_hosts-file/

Would be nice for this roslaunch feature to work more 'out of the box' for linux

dirk-thomas, 2015-05-04:
As mentioned in the question on answers.ros.org `roslaunch` already support the environment variable `ROSLAUNCH_SSH_UNKNOWN`. If you set it `roslaunch` will connect to an unknown host without complaining. So if you want to connect to hosts which are not in your known hosts lists already that is the way to go.

`roslaunch` does not use this setting by default since it want the user to actively enable connecting to unknown hosts rather then do it automatically.

closed 2015-05-04 by dirk-thomas

progtologist, 2017-01-13:
@dirk-thomas I am resurrecting this issue as I believe it has not received the attention it deserved.
The problem with remote roslaunch lies with paramiko not being able to read the known_hosts file correctly as mentioned [here](https://github.com/paramiko/paramiko/issues/243). It's not really a question of connecting to an unknown host, it's about incorrectly identifying a known host as an unknown one.

dirk-thomas, 2017-01-13:
Can you please provide more information about how to reproduce the problem.

tfoote, 2017-01-13:
There are 3 HostKeyAlgorithms ssh-rsa, ssh-dss, and ssh-ecdsa. For each of these keys there can be an entry for a host in the known_hosts file. 

The new default for command line ssh is to use the ecdsa key by default. However paramiko does not support ecdsa host keys. And cannot use that, so if you only have ecdsa host keys in your known_hosts (which is the default that ssh will drop there) paramiko cannot connect over it's list of valid transports. (rsa or dsa) 

For roslaunch(using paramiko) to work you must make sure that you have ssh-rsa host keys in your known hosts file. To do that you can just use ssh-keyscan. If you run it and look at the output you'll see that it adds 3 types by default ecdsa-sha2-nistp256, ssh-rsa, and ssh-ed255519 ( at least on my xenial box) one you do that roslaunch/paramiko should be happy.

arogowiec, 2017-09-14:
Hi!

Unfortunately I'm still facing this issue. Even after removing `.ssh/known_hosts` file and adding entry to my remote machine with command `ssh -o HostKeyAlgorithms='ssh-rsa' host` roslaunch still  complains about my host not to be on `known_hosts` file. 
I've run `ssh-keyscan` command as @tfoote  suggested and even though I saw all 3 mentioned types in the output roslaunch still complains. 
Please help me to resolve this problem.

tfoote, 2017-09-14:
@arogowiec This does not appear to be a bug which is what we try to track on github. Please ask for help on answers.ros.org with full details on how to reproduce your problem so that someone may be able to help you.

tfoote, 2022-05-26:
@hardikbvaghela That sounds like you don't have roslaunch in your environment on the remote machine. Please ask your question on https://answers.ros.org and make sure to describe fully how you have setup your remote machine and what your full console output when trying to run including your full command history and the outputs.