ros/ros_comm #616: Deadlock in ros::ServiceManager

xqms, 2015-05-08:
Hi,

I just discovered a deadlock that is happening quite often (once or twice a day in continuous operation) in my application.

The issue is that `ros::ServiceManager` has an internal `shutting_down_mutex_`, which is usually held by all functions operating on the manager.

In my case, the call chains (read from top to bottom) are as follows:

```
First thread:
  ros::service::call()
  ...
  ros::ServiceManager::createServiceServerLink() (acquires the shutting_down_mutex_)
  ...
  TransportTCP::initialize()
  ...
  TransportTCP::enableWrite() (tries to acquire TransportTCP::close_mutex_)

Second thread:
  TransportTCP::socketUpdate() (acquires the TransportTCP::close_mutex_)
  ...
  TransportTCP::close()
  ...
  ServiceManager::removeServiceServerLink() (tries to acquire the shutting_down_mutex_)
```

It seems to me that the `ServiceManager::shutting_down_mutex_` should only be held while checking `ServiceManager::shutting_down_`. But it's hard to tell without a comment on the mutex whether it is also supposed to protect against running `ServiceManager::shutdown()` while another method is already running. Can someone shed insight on that?

Here are the full backtraces from gdb (sadly without debug information):

```
First thread:

#0  __lll_lock_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/lowlevellock.S:135
#1  0x00007f1321329672 in _L_lock_953 () from /lib/x86_64-linux-gnu/libpthread.so.0
#2  0x00007f13213294da in __GI___pthread_mutex_lock (mutex=0x7f12cb4d7610) at ../nptl/pthread_mutex_lock.c:114
#3  0x00007f12feb80d8d in ros::TransportTCP::enableWrite() () from /opt/ros/indigo/lib/libroscpp.so
#4  0x00007f12feb272d9 in ros::Connection::write(boost::shared_array<unsigned char> const&, unsigned int, boost::function<void (boost::shared_ptr<ros::Connection> const&)> const&, bool) () from /opt/ros/indigo/lib/libroscpp.so
#5  0x00007f12feb27588 in ros::Connection::writeHeader(std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&, boost::function<void (boost::shared_ptr<ros::Connection> const&)> const&) () from /opt/ros/indigo/lib/libroscpp.so
#6  0x00007f12feb6dead in ros::ServiceServerLink::initialize(boost::shared_ptr<ros::Connection> const&) () from /opt/ros/indigo/lib/libroscpp.so
#7  0x00007f12feb8f00f in ros::ServiceManager::createServiceServerLink(std::string const&, bool, std::string const&, std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&) () from /opt/ros/indigo/lib/libroscpp.so
#8  0x00007f12feb6ee6f in ros::ServiceClient::call(ros::SerializedMessage const&, ros::SerializedMessage&, std::string const&) () from /opt/ros/indigo/lib/libroscpp.so
#9  0x00007f12ea3c6f16 in call<roscpp::GetLoggersRequest_<std::allocator<void> >, roscpp::GetLoggersResponse_<std::allocator<void> > > (service_md5sum=..., resp=..., req=..., this=0x7f12ea17c4a0) at /opt/ros/indigo/include/ros/service_client.h:102
#10 call<roscpp::GetLoggersRequest_<std::allocator<void> >, roscpp::GetLoggersResponse_<std::allocator<void> > > (res=..., req=..., this=0x7f12ea17c4a0) at /opt/ros/indigo/include/ros/service_client.h:74
#11 ros::service::call<roscpp::GetLoggersRequest_<std::allocator<void> >, roscpp::GetLoggersResponse_<std::allocator<void> > > (service_name=..., req=..., res=...) at /opt/ros/indigo/include/ros/service.h:71

Second thread:
#0  __lll_lock_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/lowlevellock.S:135                                                                                                                                                            
#1  0x00007f1321329672 in _L_lock_953 () from /lib/x86_64-linux-gnu/libpthread.so.0                                                                                                                                                            
#2  0x00007f13213294da in __GI___pthread_mutex_lock (mutex=0x2163228) at ../nptl/pthread_mutex_lock.c:114                                                                                                                                      
#3  0x00007f12feb8f42f in ros::ServiceManager::removeServiceServerLink(boost::shared_ptr<ros::ServiceServerLink> const&) () from /opt/ros/indigo/lib/libroscpp.so                                                                              
#4  0x00007f12feb6cc06 in ros::ServiceServerLink::onConnectionDropped(boost::shared_ptr<ros::Connection> const&) () from /opt/ros/indigo/lib/libroscpp.so                                                                                      
#5  0x00007f12feb2d524 in boost::signals2::detail::signal2_impl<void, boost::shared_ptr<ros::Connection> const&, ros::Connection::DropReason, boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (boost::shared_ptr<ros::Connection> const&, ros::Connection::DropReason)>, boost::function<void (boost::signals2::connection const&, boost::shared_ptr<ros::Connection> const&, ros::Connection::DropReason)>, boost::signals2::mutex>::operator()(boost::shared_ptr<ros::Connection> const&, ros::Connection::DropReason) () from /opt/ros/indigo/lib/libroscpp.so                                                                                                                                      
#6  0x00007f12feb28157 in ros::Connection::drop(ros::Connection::DropReason) () from /opt/ros/indigo/lib/libroscpp.so                                                                                                                          
#7  0x00007f12feb8249d in ros::TransportTCP::close() () from /opt/ros/indigo/lib/libroscpp.so                                                                                                                                                  
#8  0x00007f12feb8149a in ros::TransportTCP::read(unsigned char*, unsigned int) () from /opt/ros/indigo/lib/libroscpp.so                                                                                                                       
#9  0x00007f12feb25e32 in ros::Connection::readTransport() () from /opt/ros/indigo/lib/libroscpp.so                                                                                                                                            
#10 0x00007f12feb845c7 in ros::TransportTCP::socketUpdate(int) () from /opt/ros/indigo/lib/libroscpp.so                                                                                                                                        
#11 0x00007f12febb0efe in ros::PollSet::update(int) () from /opt/ros/indigo/lib/libroscpp.so                                                                                                                                                   
#12 0x00007f12feb55905 in ros::PollManager::threadFunc() () from /opt/ros/indigo/lib/libroscpp.so                                                                                                                                              
#13 0x00007f12fe433a4a in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0                                                                                                                                                       
#14 0x00007f1321327182 in start_thread (arg=0x7f12fa48f700) at pthread_create.c:312                                                                                                                                                            
#15 0x00007f132105400d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

dirk-thomas, 2015-05-08:
The `shutting_down_mutex_` is protecting the access to the `shutting_down_` flag. And in the creation methods as well as the shutdown method it keeps that lock until the method has finished which is the "right" behavior imo.

The problem seems to be the lock order. One path locks A and then tries to acquire B while the other locks B and then tries to acquire A. That is a classic problem of inconsistent lock hierarchies. I am not sure though how it can be changed to use the same lock order in both cases.

xqms, 2015-05-08:
> The shutting_down_mutex_ is protecting the access to the shutting_down_ flag. And in the creation methods as well as the shutdown method it keeps that lock until the method has finished which is the "right" behavior imo.

If that is the case, there is a very easy fix: Just scope the locks properly so that only the check on `shutting_down_` is protected. Currently we have:

```
ServiceServerLinkPtr ServiceManager::createServiceServerLink(const std::string& service, bool persistent,
                                             const std::string& request_md5sum, const std::string& response_md5sum,
                                             const M_string& header_values)
{
  boost::recursive_mutex::scoped_lock shutdown_lock(shutting_down_mutex_);
  if (shutting_down_)
  {
    return ServiceServerLinkPtr();
  }

  [ a lot of work and calls, which create the deadlock. ]
}
```

Is it okay to put a scope around the lock and the `if` statement? That was my question above.

dirk-thomas, 2015-05-08:
That would certainly break the protection the mutex is providing. If the `createServiceServerLink` method would release the lock after the check the `shutdown` method could run. And then after the shutdown function has completed the `createServiceServerLink` could continue with the rest of the function body leaving the service manager in an inconsistent state.

xqms, 2015-05-08:
Okay, that's what I expected. We have the same situation on the other side in `TransportTCP` then.
A comment on the mutex about what it is actually protecting might be nice ;-)

As you said the real problem is the locking order. I'd say it's not a good idea to let the low-level `TransportTCP` hold a mutex while calling the high-level `ServiceManager` object.

If you take a look at https://github.com/ros/ros_comm/blob/8713a1e22f69535a88b1a2f9c3b6a4c73abf6cf1/clients/roscpp/src/libros/transport/transport_tcp.cpp#L672 , do we need to hold the close_mutex_ while calling the callback? Any operations on the `TransportTCP` instance would acquire the lock anyway.

Moving that call out of the scope would break the deadlock, I think.

dirk-thomas, 2015-05-08:
I agree that calling the high-level function while having the `close_mutex_` locked is "wrong". But again, you can't just reduce the locking scope here. The callback is supposed to read from the socket. If the mutex does not cover that a concurrent thread can trigger the close of the connection and the socket gets cleaned up while the read is about to happen.

xqms, 2015-05-08:
> The callback is supposed to read from the socket.

But to do that, it will use `TransportTCP::read()`, which will lock the mutex again, won't it? 

> a concurrent thread can trigger the close of the connection and the socket gets cleaned up while the read is about to happen.

If `closed_` changes in the meantime, the read that is "about to happen" will do nothing. (see `TransportTCP::read()`)

`TransportTCP::close()` uses the same strategy (calling the `disconnect_cb_` callback outside of the locked scope), so I think we should mirror that.

xqms, 2015-05-08:
Actually, `read()` only locks the mutex for checking `closed_` and releases it afterwards. I think this is wrong, since a thread calling `close()` can close the socket while we are reading it.

But that is another bug (which is maybe hidden by this one).

adegroote, 2017-03-28:
Any news about it or change to fix it ?

xqms, 2017-03-28:
Sadly, I did not have time to look at this again. @adegroote did you see such a deadlock recently?

adegroote, 2017-03-28:
Well, I'm not sure. I was tracking some deadlocks, and I want to use -fsanitize=thread to track them. For now, it mostly track the deadlock you see in transport tcp, so I think it would be good to fix that first.

xqms, 2017-03-28:
You could try my fix in #616, but as mentioned there it still has an issue on `close()`.

kingsimba, 2022-10-09:
I run into the same deadlock. The callstack is exactly the same.

In my case, it will happen when a node repeatedly calls a service at 1hz (but the service provider node has crashed). Sometimes it will run for one day or two, sometimes much sooner.

I wonder if there is a workaround. Perhaps creating a service client with `persistent = true` may greatly reduce the chances and make it viable for production?

kingsimba, 2022-10-09:
Or since the deadlock is caused by one thread creating link (`createServiceServerLink`) and another recycling link(`removeServiceServerLink()`. If I reduce the rate of service call, allowing more time for recycling, the problem will not happen.

I will make some experiment and post the result when done.

closed 2025-05-31 by sloretz