ros/ros_comm #618: rosbag python: passing empty list to topics in read_messages reads all topics

dlaz, 2015-05-27:
I'm not sure if this is a bug or by design, but I find it unintuitive:

`bag.read_messages(topics=[])` will read all topics in the bag. I would expect it to read none. If this is by design, it would be helpful to make a note in the docstring.

dirk-thomas, 2015-05-28:
This is the currently intended behavior. The doc block (https://github.com/ros/ros_comm/blob/ace5b9c11af766564441bad2cb08f5fca5b73784/tools/rosbag/src/rosbag/bag.py#L242-L243) says:

* read messages from the bag
* optionally filtered by topic / topics

Hence if you don't specify topics explicitly no filtering is happening and you get all topics. And empty list is interpreted the same as the default value `None` (since both are boolean False).

I see why that seems to be weird when passing in an empty list. Its probably best to change your code to not invoke the function if the likely programmatically constructed list is empty.

dlaz, 2015-05-28:
I would argue that passing an empty list does count as explicitly specifying topics, the set of topics specified just happens to be empty. From a design standpoint, I don't think it makes sense for the meaning of the topics arg list to change based on its contents. That said, changing the behavior would clearly be a backwards incompatible change with little gain. I can submit a PR adding a brief clarification to the `@param topics` doc line.

dirk-thomas, 2015-05-28:
I would be happy with a PR which:
* either changes the logic to handle an empty list of topics differently then the default value `None` (nobody should rely on that behavior in the first place)
* or updates the docblock to state the implication when passing an empty list clearly

dlaz, 2015-08-04:
I've been super busy lately, but when I have time, I'll submit a PR which changes how empty lists are handled (and update the docblock accordingly).

closed 2015-08-26 by dirk-thomas