ros/ros_comm #619: Truncated message definitions in handshakes

athy, 2015-05-28:
When a message definition contains a sequence of =, the handshake when connecting to a topic will be invalid. If we take for instance the following message:

moveit_msgs/MoveGroupActionGoal.msg:
```
# ====== DO NOT MODIFY! AUTOGENERATED FROM AN ACTION DEFINITION ======

Header header
actionlib_msgs/GoalID goal_id
MoveGroupGoal goal
```

The field "message_definition" of the handshake when connecting a publisher and a subscriber will be
`"# "` (with a space) instead of the whole message definition. Here the message definition is truncated at `========`.

I saw this behavior on rostopic and rospy but my guess is that it is present on the whole stack (otherwise it would have been noticed earlier). My tests were done on hydro, I'm not sure about other releases.

This causes problems of interoperability with rosjava (and most probably any other implementations). Since the message definitions given during handshake have to be identical, rosjava is unable to publish/subscribe to topics with this kind of messages.

My guess is that the parser to extract the message definition just stops when it encounters a sequence of `=`. If you look at the message definition in the header for MoveGroupActionGoal, you can see that the different messages are separated by a line of equal signs:

include/moveit_msgs/MoveGroupActionGoal.h:
```
...
template<class ContainerAllocator>
struct Definition< ::moveit_msgs::MoveGroupActionGoal_<ContainerAllocator> >
{
  static const char* value()
  {
    return "# ====== DO NOT MODIFY! AUTOGENERATED FROM AN ACTION DEFINITION ======\n\
\n\
Header header\n\
actionlib_msgs/GoalID goal_id\n\
MoveGroupGoal goal\n\
\n\
================================================================================\n\
MSG: std_msgs/Header\n\
# Standard metadata for higher-level stamped data types.\n\
# This is generally used to communicate timestamped data \n\
# in a particular coordinate frame.\n\
# \n\
# sequence ID: consecutively increasing ID \n\
uint32 seq\n\
#Two-integer timestamp that is expressed as:\n\
# * stamp.secs: seconds (stamp_secs) since epoch\n\
# * stamp.nsecs: nanoseconds since stamp_secs\n\
# time-handling sugar is provided by the client library\n\
time stamp\n\
#Frame this data is associated with\n\
# 0: no frame\n\
# 1: global frame\n\
string frame_id\n\
\n\
================================================================================\n\
MSG: actionlib_msgs/GoalID\n\
# The stamp should store the time at which this goal was requested.\n\
# It is used by an action server when it tries to preempt all\n\
# goals that were requested before a certain time\n\
time stamp\n\
...
```

dirk-thomas, 2015-05-28:
From a short look at the message generators they seem to be consistent:

* `genmsg` generates the full recursive message test by splitting messages with 80 # characters on a separate line: https://github.com/ros/genmsg/blob/66589fff32e3a847646f4172cade89382a7a28ea/src/genmsg/gentools.py#L144
* `genpy` splits the string correctly on line consisting of 80 # characters (which must have a newline directly before and after): https://github.com/ros/genpy/blob/b7a1c8fed043e2aef2f21b50fa69e0b35779f2be/src/genpy/dynamic.py#L125

Also the file you mentioned (`include/moveit_msgs/MoveGroupActionGoal.h`) clearly contains the correct full `value`.

Can you please provide a short, self contained, correct example (SSCCE) which demonstrates the issue.

dirk-thomas, 2016-03-17:
Closing due to lack of response.

closed 2016-03-17 by dirk-thomas