ros/ros_comm #620: rosbag play --topics is broken

mintar, 2015-06-03:
On Indigo, I couldn't find a way to get `rosbag play --topics` to work. IIRC, on Hydro, the following used to work, but produces an error on Hydro:

    $ rosbag play mybag.bag --clock --topics /mytopic
    [ERROR] [1433321952.976544128]: Error reading options: When using --topics or --pause-topics, --bags should be specified to list bags.
________________________________________________________________________________

Ok, so I tried to humor rosbag:

    $ rosbag play --clock --topics /mytopic --bags mybag.bag  
    Usage: rosbag play BAGFILE1 [BAGFILE2 BAGFILE3 ...]
    
    rosbag: error: You must specify at least 1 bag file to play back.
________________________________________________________________________________

Last try:

    $ rosbag play mybag.bag --clock --topics /mytopic --bags mybag.bag
    [ INFO] [1433321989.060645972]: Opening mybag.bag
    [ INFO] [1433321989.061010974]: Opening mybag.bag
    
    Waiting 0.2 seconds after advertising topics... done.
    
    Hit space to toggle paused, or 's' to step.

This doesn't produce an error, but of course now every message is replayed twice. Did I miss something obvious, or is this a regression from Hydro?

mintar, 2015-06-04:
Update: Just confirmed that version 1 used to work in Hydro.

Versions:
* hydro: rosbag 1.10.12 (works)
* indigo: rosbag 1.11.13 (broken)

mintar, 2015-06-04:
Update: This was introduced in b7796fea : in [line 238](https://github.com/ros/ros_comm/blob/b7796fea666aac7b330aad870f7e7b994eda8daf/tools/rosbag/src/rosbag/rosbag_main.py#L238), setting the --bags arg was removed.

mintar, 2015-06-04:
So, in summary the problem is this. This is how it worked in hydro:

1. are positional args for bag files present? **yes** [(here)](https://github.com/ros/ros_comm/blob/d13d0263fa632e97500a81bf16250ab0d32cb169/tools/rosbag/src/rosbag/rosbag_main.py#L135-L136)
2. is `--topics` set? **yes** => add `--bags` to args [(here)](https://github.com/ros/ros_comm/blob/d13d0263fa632e97500a81bf16250ab0d32cb169/tools/rosbag/src/rosbag/rosbag_main.py#L224)
3. is `--topics` set, but not `--bags`? **no** (otherwise exception) [(here)](https://github.com/ros/ros_comm/blob/d13d0263fa632e97500a81bf16250ab0d32cb169/tools/rosbag/src/play.cpp#L135-L136)

In  b7796fe, number (2) was removed, causing this bug.

So, what to do about this?

1. restore the old behavior by adding `--bags` if `--topics` (and now presumably `--pause-topics`) is present, or
2. don't require positional args if `--bags` is present?

dirk-thomas, 2015-06-09:
Fixed with #624

closed 2015-06-09 by dirk-thomas

v4hn, 2015-08-19:
Hey guys,

would be great to see this fix in a release!
Until now --topics is still broken in the latest release.

mintar, 2015-11-09:
The fix has still not been released into Indigo. Could we get another release please?

dirk-thomas, 2015-11-09:
Th patch has been released a while ago and is sitting in shadow-fixed for a soak time. Due to other regressions it will sit there a bit more until it can be synced to the public repo.

andrewcapodieci, 2016-12-09:
Just adding my voice in here...I'd like to see this fix in release as well. Thanks.

nicolaerosia, 2017-10-05:
this seems to be broken on Lunar:
```
user@ubuntu1704:~/bagfiles$ rosbag play --clock --topics /mytopic 2017-10-05-16-53-43.bag 
Usage: rosbag play BAGFILE1 [BAGFILE2 BAGFILE3 ...]

rosbag: error: You must specify at least 1 bag file to play back.
```

dirk-thomas, 2017-10-05:
@nicolaerosia A recent regression has been fixed in #1156 which will be available with the next patch release.

nicolaerosia, 2017-10-06:
@dirk-thomas I should have searched more, thanks!