ros/ros_comm #621: No NodeHandle::getParam() overload for float?

theidealist, 2015-06-03:
[Moved report from: https://github.com/ros/roscpp_core/issues/35]
I'm not sure if this is the correct project, but it seems like the correctly named one and I could not find which repo actually has node_handle.h.

In my version of ros (indigo), NodeHandle does not have a ::getParam() overload for float, the rationale for which is not clear. This seems like a bug/oversight.

Thank you!

dirk-thomas, 2015-06-03:
The parameter server only stores double values - never float values. Therefore the `NodeHandle` API only exposes `getParam()` with `float` and leaves the conversion (with potential precision loss) to the user.

The choice for not storing floats natively on the parameter server is related to cross language support: http://answers.ros.org/question/10938/why-cant-you-use-floats-for-accessing-parameters-in-roscpp/

Based on that I don't think the function is missing is not an oversight but intended.

theidealist, 2015-06-03:
Can I ask why it has an overload for std::vector<float> then?

dirk-thomas, 2015-06-03:
I guess since it is more effort to convert a vector of double to float that method is provided as syntactic sugar.

theidealist, 2015-06-03:
Can we convert this to a request then for equivalent syntactic sugar in the single case?  For example, anything using ::param() in a templated manner runs into a compile time error here if ever called with a float. It seems like a small thing to add and is certainly no less safe than the public API already is through `std::vector<float>`.

I could do a patch for it pretty quick myself if accepted?

theidealist, 2015-06-03:
Submitted pull request 622 for review and consideration:
https://github.com/ros/ros_comm/pull/622

dirk-thomas, 2015-06-04:
I will bring this up in our next meeting for discussion. I am a bit worried that the argument could also then be used to add functions for various other types: `short`, `long`, `unsigned *`

And I think the original idea was to intentionally not provide API for those.

theidealist, 2015-06-04:
Thank you, Dirk.  I'll look forward to hearing what you all come up with.  Note that I am not requesting support for alternative integer types, just consistent support of floats via "syntactic sugar".  Let requests for the other types come from your other users ;-)

I noticed in commit 54a46df2 there is a fix very similar to the one I implemented in PR 622, which I think further motivates exposing floats at the NodeHandle level.  But, after that realization, it seems that my PR should probably be modified to just make use of the float overloads that are already in the param namespace directly.  Again, let me know and I'll be happy to make whatever adjustments might be necessary.

Thanks again.

dirk-thomas, 2015-07-01:
Addressed by #635.

closed 2015-07-01 by dirk-thomas