ros/ros_comm #629: get_topic_class from rostopic returns wrong real topic name

awesomebytes, 2015-06-16:
Hello, from the description (and the usage I found of it) of **get_topic_class** of the rostopic module (https://github.com/ros/ros_comm/blob/indigo-devel/tools/rostopic/src/rostopic/__init__.py#L449-L463) I think there may be a bug in it.

Given a topic that publishes a message with subfields, in this example a `PoseStamped` message in the topic name `/posestamped`, if I do `get_topic_class('/posestamped')` it will return as second return value the "real topic name" and it will correctly be **'/posestamped'**. But if I do `get_topic_class('/posestamped/pose')` which would be the subfield of message type `Pose` so I believe it's "real topic name" would be '/posestamped'... but I get as "real topic name" **'/posestamped/pose'**. Furthermore if I do `get_topic_class('/posestamped/pose/position/x')` it will return as "real topic name" **'/posestamped/pose'**.

Which I still believe it's wrong. But maybe my interpretation is wrong.

This standalone snippet of code exemplifies the problem, also I reproduce the code here:
https://gist.github.com/awesomebytes/362f5a31a846952068c0

```python
import rospy
from rostopic import get_topic_class
from geometry_msgs.msg import PoseStamped

TESTING_TOPIC = '/posestamped'

if __name__ == '__main__':
    rospy.init_node('test_get_topic_class', anonymous=True)

    # First we need a publisher that publishes some class that has subfields
    # I like PoseStamped, as it has a header and also some subfields
    ps_pub = rospy.Publisher(TESTING_TOPIC, PoseStamped, queue_size=1, latch=True)

    # We need it publishing, so lets loop publishing, and check if get_topic_class correctly
    # gives us the topic name as it's docstring states:
    #   Get the topic message class
    #   :returns: message class for topic, real topic
    #   name, and function for evaluating message objects into the subtopic

    # Publish a message
    ps_pub.publish(PoseStamped())
    # Wait for some time to let the universe get comfy
    rospy.sleep(0.1)

    # Let's check the "real topic", after the print line you can see the output given
    _, real_topic, _ = get_topic_class(TESTING_TOPIC)
    print "get_topic_class says real_topic of '" + TESTING_TOPIC + "' is: '" + real_topic + "'"
    # get_topic_class says real_topic of '/posestamped' is: '/posestamped'

    SUBFIELD_HEADER = TESTING_TOPIC + '/header'
    _, real_topic, _ = get_topic_class(SUBFIELD_HEADER)
    print "get_topic_class says real_topic of '" + SUBFIELD_HEADER + "' is: '" + real_topic + "'"
    # get_topic_class says real_topic of '/posestamped/header' is: '/posestamped/header'

    SUBFIELD_POSE = TESTING_TOPIC + '/pose'
    _, real_topic, _ = get_topic_class(SUBFIELD_POSE)
    print "get_topic_class says real_topic of '" + SUBFIELD_POSE + "' is: '" + real_topic + "'"
    # get_topic_class says real_topic of '/posestamped/pose' is: '/posestamped/pose'

    SUBFIELD_POSE_POSITION = TESTING_TOPIC + '/pose/position'
    _, real_topic, _ = get_topic_class(SUBFIELD_POSE_POSITION)
    print "get_topic_class says real_topic of '" + SUBFIELD_POSE_POSITION + "' is: '" + real_topic + "'"
    # get_topic_class says real_topic of '/posestamped/pose/position' is: '/posestamped/pose'

    SUBFIELD_POSE_POSITION_X = TESTING_TOPIC + '/pose/position/x'
    _, real_topic, _ = get_topic_class(SUBFIELD_POSE_POSITION_X)
    print "get_topic_class says real_topic of '" + SUBFIELD_POSE_POSITION_X + "' is: '" + real_topic + "'"
    # get_topic_class says real_topic of '/posestamped/pose/position/x' is: '/posestamped/pose'

    # Leaving the process open for debugging purposes
    rospy.spin()```

I have no idea if this is the expected behaviour, and if it is, why... and what alternative could I have to have the real-"real topic name". Thank you for your time!

awesomebytes, 2015-06-16:
Meanwhile I've implemented this `get_original_publisher_topic` (Standalone snippet that exemplifies it's use also):
https://gist.github.com/awesomebytes/6c3acc945fc839da5ca1

```python
def get_original_publisher_topic(topic):
    master = rosgraph.Master('/rostopic')
    try:
        state = master.getSystemState()
        pubs, subs, _ = state
        pubs_that_start_with_topic = []
        for pub_name, other_info in pubs:
            # Check if publisher starts with the same name than the provided topic
            if topic.startswith(pub_name):
                pubs_that_start_with_topic.append(pub_name)
        if len(pubs_that_start_with_topic) == 0:
            return None
        else:
            # We should get the publisher with the longer name that contains this
            # The way to understand it is because if we have
            # a publisher of PoseStamped in the topic /posestamped
            # and a publisher of Pose in the topic /posestamped/pose
            # The original topic we want is /posestamped/pose as I would consider the standard
            # use of topics to subscribe to the real topic, not a subfield
            return max(pubs_that_start_with_topic)

    except socket.error:
        print("Unable to communicate with master!")
```
It may not be generic enough, but I think the idea is good. I could make a PR with a better implementation given some guidance. (If what I'm doing makes sense at all).

awesomebytes, 2015-07-01:
Any comment on this?

dirk-thomas, 2015-07-02:
I did not have any time to look at your code and it might take a while until I can get to it. Sorry.

cwecht, 2019-03-22:
I was not able to reproduce this with melodic. @awesomebytes could you check, if this is still an issue?

dirk-thomas, 2019-03-22:
Based on the latest comment I will go ahead and close the ticket for now. Please feel free to comment if this is still a problem and the ticket can be reopened.

closed 2019-03-22 by dirk-thomas