ros/ros_comm #630: need non blocking spin(async=True, callback=myFunc)

asmodehn, 2015-06-20:
When we are using rospy not in a direct control flow, rospy.spin() is blocking.
So in an event based design ( reversed control flow, waiting for callbacks ), it would be useful to have an spin(async=True, callback=myFunc) with the callback getting called whenever rospy is shutdown.
This way the calling software doesn't need to loop and check at regular interval...

asmodehn, 2015-06-20:
or we could have : 
* node_init()
* node_start_spin()
* node_stop_spin()

Basically I m using rospy with werkzeug, and signal handling is a pain to get right...

asmodehn, 2015-06-20:
I finally managed to solve my problem using some python magic.
In case someone arrives here in quest of illumination about how to disable signal handlers and still properly cleaning up everything, here is how I did it : 

```python

"""
Interface with ROS.
"""
from contextlib import contextmanager

# A context manager to handle rospy init and shutdown properly.
@contextmanager
def RostfulNode(ros_args):
    #we initialize the node here, passing ros parameters.
    #disabling signal to avoid overriding callers behavior
    rospy.init_node('rostful', argv=ros_args, anonymous=True, disable_signals=True)
    rospy.logwarn('rostful node started with args : %r', ros_args)

    class RostfulNodeImpl(object):
        def __init__(self):
              # lets get ready
              myparam = rospy.get_param('~myparam', [])
              pass
        def doAwesomeROSstuff(self):
              # just do it
              pass

    yield RostfulNodeImpl()
    rospy.logwarn('rostful node stopped')
    rospy.signal_shutdown('Closing')

```

and the caller : 
```python
        with rostful_node.RostfulNode(ros_args) as node:
            self._setup(node)
            # no need to spin() or block on anything

            rostful_server.app.logger.info('Starting Flask server on port %d', port)
            # debug is needed to investigate server errors.
            # use_reloader set to False => killing the ros node also kills the server child.
            rostful_server.app.run(host=host, port=port, debug=True, use_reloader=False)

        #RostfulNode gets cleaned up and rospy.signal_shutdown() is called when we get out of scope.
```

dirk-thomas, 2015-06-20:
I am not sure if I understand your use case. Couldn't you simply do the following using the existing API:
* fork a new thread
* call `spin()` in there
* when ever the thread finishes (which means ros shut down) trigger custom behavior

asmodehn, 2015-06-21:
My usecase is a ros_init() call inside a WSGI App.
Feel free to have a look at this commit if you want to investigate : 
https://github.com/asmodehn/rostful/commit/c6a725f07e86c43934d124d25154d55a1bda8a4c
Basically if I call ros_init inside the wsgi app, and I kill the webserver(SIGINT), the rosnode is not killed immediately (needs to go to SIGTERM)... probably because I didnt spin (I don't want to block), and therefore I didnt call signal_shutdown.

The new thread is what I was thinking about at first, and the idea behind the async_spin. But Triggerring custom behavior also includes passing callback into a queue and polling it from the main thread... a lot of code when I realized that rospy.spin() only does : 
̀̀̀```python
try:
        while not rospy.core.is_shutdown():
            rospy.rostime.wallsleep(0.5)
    except KeyboardInterrupt:
        logdebug("keyboard interrupt, shutting down")
        rospy.core.signal_shutdown('keyboard interrupt')
```

so the only interesting part for my usecase is to catch the KeyboardInterrupt and call signal_shutdown.

Since the KeyboardInterrupt is already caught by the webserver, and kill the wsgiapp(and delete the node), I just need to make sure I call the signal_shutdown when this happens, to avoid blocking during the shutdown.
Managing resources is the purpose of the with operator if I understood properly, and indeed it works, and require less code than having another thread, a callback queue, etc, just to be able to catch an exception.

My new implementation is there : https://github.com/asmodehn/rostful/commit/aa4c12b6fe405adbc1a286ac8222767e07cc08f2

So async spin is not needed. and maybe spin isn't either.
But if we have to spin, I just think rospy should probably provide a non-blocking spin as well...

wjwwood, 2015-06-21:
As you've stated, the `rospy.spin()` just waits for the the shutdown event. It's just a convenient way to block while asynchronous middleware stuff is happening. This doesn't at all line up with `roscpp` where `spin`ing is necessary for middleware activity to occur, but it is the namesake for this function in rospy. The `rospy.spin` function would be more appropriately called `rospy.wait_for_shutdown`.

So, since `rospy.spin` doesn't do anything but block on shutdown, you can safely ignore it rather than look for an async version of `rospy.spin`. If you want a call back on shutdown, you can probably use `rospy.core.add_shutdown_hook`, see:

http://docs.ros.org/api/rospy/html/rospy.core-module.html#add_shutdown_hook

asmodehn, 2015-06-21:
I see... thanks !

dirk-thomas, 2015-06-22:
Based on the discussion I will close this and mark it "wontfix". Please feel free to comment on it with more information.

closed 2015-06-22 by dirk-thomas