ros/ros_comm #638: More customization allowable relay command line tool

wkentaro, 2015-07-15:
I'd like to relay topics with changes of msg type like below.
```
$ rosrun rospy_tutorials talker

# THIS IS MAIN >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
$ rosrun topic_tools relay_field /chatter /image sensor_msgs/Image "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: 'm.data'
height: 0
width: 0
encoding: ''
is_bigendian: 0
step: 0
data: ''"
# THIS IS MAIN <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

$ rostopic echo /image
 header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: 'hello world 1436966790.72'
height: 0
width: 0
encoding: ''
is_bigendian: 0
step: 0
data: ''
...
```
Is this kind of feature already implemented in topic_tools of ros_comm?
Or should I implement this as new tool?

wjwwood, 2015-07-15:
It is not a tool we already have as far as I know.

dirk-thomas, 2015-09-16:
New tool implemented in #639.

closed 2015-09-16 by dirk-thomas