ros/ros_comm #643: Proposal: Docker interface for roslaunch

ruffsl, 2015-07-23:
TL;DR Proposal to run ROS nodes in containers using roslaunch

## Background
ROS provides users and developers a common platform in witch to build their robotic applications, share common code and packages, and collaborate on multidisciplinary software challenges. As ROS provides a Debian based package library, this enables quick distribution and access to released code bases.

For developing the code bases themselves, ROS provides a set of build tools, such as catkin, to standardize packages and simplify the compilation process of larger projects. In this way, developers can describe the build process and required dependencies that then library maintainer can use for coordinating build farms and package distribution.

## Problem Statement
For users who require the latest version of an application or for developers who need to build from source, a few problems present themselves:

* release time
  * Users who rely on latest features or collaborate with developers, but are not entirely familiar with the compilation of the project, usually must wait until the next cycle of package upgrades are released over the Debian library, or risk breaking there own deployment environment in attempting to learn to build multiple components together from source.
* dependencies
  * Even for experienced developers that may be collaborating, building packages from source may not treat it as a trivial task due to dependency and version conflicts, environment specifics and other miscellaneous gotchas. So sharing and testing works in progress can be taxing when developing on more than one component.
* RR&D
 * Without relying on Debian packages, Repeatability, Reproducibility and Deployablity become a concern when debugging for development or shipping for production. So an individual's mileage in building and running a package on there own system may very.

## Proposal
Provide support for launching ROS nodes within docker containers using roslaunch. The idea is that this would mimic how the `machine` tag works currently, enabling user to specify the remote machine to run the specified launch configuration. Similarly, a `container` tag could be used to allow the user to specify that an encompassing section of a launch file should be invoked inside the described container.

### Challenges
In its current state, roslaunch is rather complex and fragile with minimal test coverage. Adding enough machinery to support docker integration without regression would remain a delicate task. Other specifics include:

* use behavior
 * How should such an implementation be expected to work? 
 * What behavior or defaults should be perscribed to enable users to containerize their nodes?
 * What does the user specify, the image, tag, compose file?
 * How should the networking specifics be provided?
* dependencies
 * Including docker features into roslaunch may introduce dependencies for the docker python api, [docker-py](https://github.com/docker/docker-py). Are there simply fall-back behaviors for users who may not have the docker dependencies installed?
* orchestration
 * Is this best direction to take with coordinating ROS and containers? At what level should ROS and docker interact? Should it be the users responsibility to manage containers, with ros launch simply invoked inside a container? Should a separate tool be formed to be the intermedary?

## Summary
With the growth of the container ecosystem, it is important that ROS provides a degree of interfaces for containerization and enable users to leverage the RR&D benefits they provide. To do this, a feature for defining how run ROS nodes with containers using launch files is proposed.

tfurf, 2016-07-06:
A year late, but +1 from me.

IanTheEngineer, 2016-07-06:
Agreed! +1 from me too. @ruffsl did you get to implement any of this functionality on a fork somewhere?

tfurf, 2016-07-06:
After I saw this the first time (and +1ed it), I started looking at the rest of the [roslaunch CLI arguments](http://wiki.ros.org/roslaunch/Commandline%20Tools). My thoughts so far are that roslaunch is a fragile beast, and once I solidify a configuration, I really hestitate to touch it, and would rather just spin up a set of nodes (as a service) in Docker, via Compose (usually). Because the CLI can give you the specific options for a node, resolved from a set of launch files (linked via inclusion), you could do (based on ROS example from hub):

```yaml
version: '2'
...
  talker:
    build: .
    container_name: talker
    environment:
      - "ROS_HOSTNAME=talker"
      - "ROS_MASTER_URI=http://master:11311"
    command: rosrun $(roslaunch --args talker roscpp_tutorials test.launch)
...
```

This is sort of OK, not pretty to me, but it kind of works. What is missing for me, is the ability to use roslaunch to run maybe a subset of a total launch environment, so that the 'atomic' processes can all be near each other (a single container/service). Almost like a namespace (ish) per container. This sort of goes against the Docker grain a little bit, but also lends itself to isolating the 'services' that a ROS environment ends up providing.

ruffsl, 2016-07-06:
I've somewhat paused on this front, and don't have any working prototype with this functionality. Although I like the idea of the integrated functionality afforded, I'm now seeing the greater challenges in design and maintenance. 

The Docker API is perhaps more stable than it was a year ago, but still a lot is changing and improving underneath us. I think it may take a fair amount of continual effort to keep roslaunch interface compatible and relevant. Exposing the scope of parameters and setting to uses would also be tricky and perhaps always behind the curve compared to dedicated tooling such as Docker Compose would provide. Even just enabling users to leverage 3rd party plugins would be a challenge to accomodate, such as  [Nvidia's Docker plugin](https://github.com/NVIDIA/nvidia-docker), a plugin I frequently utilize for deep learning tasks. 

In some ways I guess I'm now seeing roslaunch as a pure means to launch ROS, and docker a means to launch containers, and so putting one ahead of the other or splitting roles may be like forcing the Atlantis to lift the STS-135's SRB (booster) off the launchpad.

maxpfingsthorn, 2016-07-08:
I would agree that roslaunch does its job well to launch nodes, and anything on a higher level (e.g. running things on remote hosts, containers, VMs, etc) should be using roslaunch underneath. Using the built-in orchestration features of Docker is definitely a good idea. I would favor one container per (small) launch file, only including the relevant packages, with the default command starting that launch file.

What seems to be missing is something to conveniently build containers from built workspaces and some indication which launch file to run. As well as something to conveniently build workspaces in containers in the first place, especially for development. Maybe some of the work on the Docker-based ROS buildfarm would be relevant here.

ruffsl, 2018-05-09:
cc @BillWSY

closed 2025-05-31 by sloretz