ros/ros_comm #651: Docstring for rospy.on_shutdown() is wrong

opqdonut, 2015-08-20:
The docstring claims that the hook should take no arguments, but in the "already shutdown" case the hook gets called with one argument.

The call is here: https://github.com/ros/ros_comm/blob/ace5b9c11af766564441bad2cb08f5fca5b73784/clients/rospy/src/rospy/core.py#L316

spaghetti-, 2015-08-24:
I think we can just take out the argument. It is kind of difficult to work around this otherwise.

wjwwood, 2015-09-01:
@opqdonut I do not believe that is true. AFAIKT, those callbacks do take a string argument:

- https://github.com/ros/ros_comm/blob/ace5b9c11af766564441bad2cb08f5fca5b73784/clients/rospy/src/rospy/core.py#L360
- https://github.com/ros/ros_comm/blob/ace5b9c11af766564441bad2cb08f5fca5b73784/clients/rospy/src/rospy/core.py#L374

opqdonut, 2015-09-02:
the docstring for on_shutdown claims otherwise, as I said in my first comment :)

https://github.com/ros/ros_comm/blob/ace5b9c11af766564441bad2cb08f5fca5b73784/clients/rospy/src/rospy/client.py#L109

wjwwood, 2015-09-02:
Probably related to this:

- https://github.com/ros/ros_comm/commit/cdf44f3b103a951dbde25f8b6a4066a4afb40e82
- https://github.com/ros/ros_comm/issues/586

dirk-thomas, 2015-09-17:
The function `client.py:on_shutdown()` (https://github.com/ros/ros_comm/blob/ace5b9c11af766564441bad2cb08f5fca5b73784/clients/rospy/src/rospy/client.py#L109-L116) registers a **client** shutdown function (https://github.com/ros/ros_comm/blob/ace5b9c11af766564441bad2cb08f5fca5b73784/clients/rospy/src/rospy/core.py#L343-L352).

Those are supposed to be always called without arguments: https://github.com/ros/ros_comm/blob/ace5b9c11af766564441bad2cb08f5fca5b73784/clients/rospy/src/rospy/core.py#L396-L399

Therefore I would argue the current docstring is correct and does not need to be modified.

But when a callback is registered after shutdown it tries to pass an argument (https://github.com/ros/ros_comm/blob/ace5b9c11af766564441bad2cb08f5fca5b73784/clients/rospy/src/rospy/core.py#L316). Therefore I would propose to not pass that argument if the function does not accept an argument. Please see #662 for the resulting patch.

If some of you could please try it in scenarios where it failed before that would be great to confirm that it works as intended.

commit 9fa958b3 (2015-09-18): fix signaling already shutdown to client hooks with the appropriate signature (fix #651)
closed 2015-09-18 by dirk-thomas