ros/ros_comm #658: catkin does not allow paths with special characters

VictorLamoine, 2015-08-29:
Creating a work-space in a path containing one or more special characters will lead to errors.

- First example, work-space contains a `é` (catkin_workspac**é**)
```bash
[roslaunch.config][INFO] 2015-08-29 13:47:15,356: loading config file /home/victor/catkin_workspacé/src/fanuc_experimental/fanuc_r1000ia_support/launch/test_r1000ia80f.launch
[roslaunch][ERROR] 2015-08-29 13:47:15,381: 'ascii' codec can't decode byte 0xc3 in position 28: ordinal not in range(128)
```

- Second example, work-space contains a `space` character
```bash
~/SpiderOak Hive/catkin_workspace/$ roslaunch fanuc_m10ia_support find: `/home/dell/SpiderOak': No such file or directory find: `Hive/catkin_workspace/src/fanuc/fanuc_m10ia_support': No such file or directory
```

[Related ROS-Answers question](http://answers.ros.org/question/216626/how-do-i-use-catkin-in-a-directory-containing-a-space-character/?comment=216731).

dirk-thomas, 2015-08-29:
Please provide a reproducible example. E.g. a rosinstall files describing the repos you use in your workspace as well as a command sequence to trigger the problems.

From a first look neither of the problems seems to be related to `catkin` but to `roslaunch`.

VictorLamoine, 2015-08-30:
Ubuntu 14.04
Bash terminal
`ros-indigo-roslaunch` version `1.11.13-0trusty-20150522-1835-+0000` (`amd64`)

```bash
$ cd ~ && mkdir "catkin workspace" && cd catkin\ workspace/
$ mkdir src && cd src/ 
$ git clone https://github.com/ros-industrial/industrial_core.git
...
$ git clone https://github.com/ros-industrial/fanuc.git
...
$ catkin_init_workspace 
Creating symlink "/home/victor/catkin workspace/src/CMakeLists.txt" pointing to "/opt/ros/indigo/share/catkin/cmake/toplevel.cmake"
$ cd .. && catkin_make
...
$ source devel/setup.bash 
$ roslaunch fanuc_m10ia_support testfind: `/home/victor/catkin': No such file or directory
find: `workspace/src/fanuc/fanuc_m10ia_support': No such file or directory
```

dirk-thomas, 2015-08-31:
Thanks for the detailed example. I was able to reproduce it based on that and created a fix for the roslaunch completion function: ros/ros#88

If you want to try the fix the easiest way is probably to apply the patch to the installed file directly (/opt/ros/indigo/share/rosbash/rosbash).

I also updated the title of this ticket to describe the actual problem more precisely.

dirk-thomas, 2015-08-31:
Can you please try if the patch also addresses the problem with special characters? I only tested white spaces.

VictorLamoine, 2015-08-31:
This completely fixes the auto completion feature, including special characters (I have tested `~è*ù`).

But this does not (completely) solve the issue; either launching `test_m10ia.launch` does not load the robot or the `.launch` won't launch at all.

This looks similar to [ros/ros/#54](https://github.com/ros/ros/issues/54):
```bash
victor@desktop:~/catkin_~è*#ù$ roslaunch fanuc_m10ia_support test_m10ia.launch 
... logging to /home/victor/.ros/log/5aebf60e-500a-11e5-9dc9-10c37b6bfb43/roslaunch-desktop-17009.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

'ascii' codec can't decode byte 0xc3 in position 21: ordinal not in range(128)
The traceback for the exception was written to the log file
```

With a work-space containing a white space (and no special characters) the `.launch` is launched, RViz shows but no robot and nothing unusual in the logs.

dirk-thomas, 2015-08-31:
Since the error message states "The traceback for the exception was written to the log file" it would have been beneficial to include this traceback from the log in the comment. Anyway I have reproduced the problem locally and had the following in the log:

    [roslaunch][ERROR] 2015-08-31 14:12:56,091: Traceback (most recent call last):
      File "<string>", line 298, in main
      File "/tmp/catkin workspacè/src/ros_comm/tools/roslaunch/src/roslaunch/parent.py", line 271, in start
        self.runner.launch()
      File "/tmp/catkin workspacè/src/ros_comm/tools/roslaunch/src/roslaunch/launch.py", line 646, in launch
        self._setup()
      File "/tmp/catkin workspacè/src/ros_comm/tools/roslaunch/src/roslaunch/launch.py", line 623, in _setup
        self._launch_core_nodes()
      File "/tmp/catkin workspacè/src/ros_comm/tools/roslaunch/src/roslaunch/launch.py", line 498, in _launch_core_nodes
        name, success = self.launch_node(node, core=True)
      File "/tmp/catkin workspacè/src/ros_comm/tools/roslaunch/src/roslaunch/launch.py", line 526, in launch_node
        process = create_node_process(self.run_id, node, master.uri)
      File "/tmp/catkin workspacè/src/ros_comm/tools/roslaunch/src/roslaunch/nodeprocess.py", line 131, in create_node_process
        args = create_local_process_args(node, machine)
      File "/tmp/catkin workspacè/src/ros_comm/tools/roslaunch/src/roslaunch/node_args.py", line 242, in create_local_process_args
        matches = roslib.packages.find_node(node.package, node.type, rospack)
      File "/tmp/catkin workspacè/src/ros/core/roslib/src/roslib/packages.py", line 400, in find_node
        return find_resource(pkg, node_type, filter_fn=_executable_filter, rospack=rospack)
      File "/tmp/catkin workspacè/src/ros/core/roslib/src/roslib/packages.py", line 500, in find_resource
        search_paths = catkin_find(search_dirs=['libexec', 'share'], project=pkg, first_matching_workspace_only=True)
      File "/tmp/catkin workspacè/src/catkin/python/catkin/find_in_workspaces.py", line 127, in find_in_workspaces
        p = os.path.join(p, project)
      File "/usr/lib/python2.7/posixpath.py", line 80, in join
        path += '/' + b
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 20: ordinal not in range(128)

The problem is that the default encoding of your Python version is `ASCII` but you are using non-ASCII characters in your file system. I am not sure of we can address that in ROS.

There are also several not ROS related compenents (e.g. using Qt via CMake) which won't handle that case. So I am not sure if it is "worth the effort".

If you would like to look into it further pull requests are definitely appreciated :wink:

VictorLamoine, 2015-09-01:
I think you are right; it's not worth the pain investigating this much further.
I don't understand why it doesn't work with a white space though, it's part of the `ASCII` table but the robot never shows.

[ros/ros/#54](https://github.com/ros/ros/issues/54) is a first step towards compatibility so it's better than nothing :+1: .
Feel free to close if you feel the need to.

closed 2015-09-16 by dirk-thomas

gmm96, 2017-03-13:
I have been suffering this issue in Kubuntu 16.04 with ros kinetic. I hadn't any special character in the path of the project, but it failed every time I tried to execute roslaunch.

Finally, I have been able to workaround this bug by adding the following lines to the file /usr/lib/python2.7/posixpath.py

```python
import sys
reload(sys)
sys.setdefaultencoding("utf-8") 
```

I know it isn't the best way to solve this issue, but it's the only thing that works for me.
Hope this can help other people.

dirk-thomas, 2017-03-13:
@gmm96 Patching a core Python file like this is pretty dangerous. I can only advice against trying this approach.

gmm96, 2017-03-13:
@dirk-thomas I know the risks I'm taking with this kind of solution, but I don't mind to be the guinea pig. If I have troubles with this patch, I will report them here.