ros/ros_comm #664: Inconsistent node name validation between roscpp and rospy

mvollrath, 2015-09-17:
This works:

    $ rosrun any_pkg any_rospy_node __name:=foo-bar

This does not:

    $ rosrun any_pkg any_roscpp_node __name:=foo-bar
    terminate called after throwing an instance of 'ros::InvalidNameException'
      what():  Character [-] at element [3] is not valid in Graph Resource Name [foo-bar].  Valid characters are a-z, A-Z, 0-9, / and _.
    Aborted (core dumped)

dirk-thomas, 2015-09-18:
It looks like that `roscpp` validates the name before using it. `rospy` might simply not perform any validation. Have you tried completely arbitrary names with Python?

mvollrath, 2015-09-19:
The first thing I noticed is that an invalid name in rospy is only a warning when passed to init_node():

https://github.com/ros/ros_comm/blob/indigo-devel/clients/rospy/src/rospy/client.py#L261-L266

Next, a name provided from the command line doesn't even get a warning because it is pulled in after that check:

https://github.com/ros/ros_comm/blob/indigo-devel/clients/rospy/src/rospy/client.py#L261-L266

So, anything that makes it through either implementation of `resolve_name()` (different ones are used for arg and cmdline, and neither validate) is accepted in rospy.  Looks like the validators in `rospy.names` are unfinished too.  The validator in `rosgraph.names` miight be appropriate for a node name:

https://github.com/ros/ros_comm/blob/indigo-devel/tools/rosgraph/src/rosgraph/names.py#L235-L244

The pattern is `^[A-Za-z][\w_]*$` (which seems like it could be simplified since `_` is in the set of `\w` according to [the docs](https://docs.python.org/2/library/re.html#regular-expression-syntax)).

roscpp init validation follows the same rules:

https://github.com/ros/ros_comm/blob/indigo-devel/clients/roscpp/src/libros/this_node.cpp#L131-L140

which calls `names::resolve()` which calls:

https://github.com/ros/ros_comm/blob/indigo-devel/clients/roscpp/src/libros/names.cpp#L56-L94

dirk-thomas, 2015-09-19:
As you have pointed out the validation behavior can clearly be improved / unified for all code paths and languages.

For already released ROS distro we can only add warnings for previously accepted names. If we would suddenly reject them that can cause regressions in existing code (since it might not follow the convention). Therefore I would suggest to only introduce stricter checks resulting warnings for now and in the next ROS release (aka k-turtle) change some of the warnings into errors.

It would be great if someone could work on a PR for this since I might not find the time to work on this change.

mvollrath, 2015-09-19:
Are acceptable node names defined in a REP or some other client standards doc?  Should the master reject bad names in the next distro?

mvollrath, 2015-09-20:
It's sort of a stretch, but there's a vulnerability here.

When shell escapes are allowed in any node identifier, a script can be circumvented to expose the value of a private environment variable.

For example:

* Start a rospy node with `__name:='$HOME'`
* Run `rosnode list` and see that `/$HOME` is in the list.
* Run `tcpdump -X dst port 11311` to see communications with the master.
* Run `rosnode info "$HOME"` (as an ignorant bash script iterating over `rosnode list` might do) and see that the value of `$HOME` is in the captured data.

roscpp's validation ensures that the name can be used as an identifier in practically any language (length aside, maybe) and shell escapes are impossible.

tfoote, 2015-09-20:
Naming rules for graph resources are defined here: http://wiki.ros.org/Names

minshallj, 2015-09-21:
Do you want pull requests for all the *-devel branches? And which branch should change the warning to an error?

dirk-thomas, 2015-09-28:
A pull request against the latest deve branch is sufficient. Changes will be backported from there after they have been merged.

PR #669 (2015-10-13): Ft rospy validate name
This will validate names passed from the command line which were previously going un-validated.

Fix #664
closed 2015-10-13 by dirk-thomas