ros/ros_comm #671: setCallBackQueue works only if set before message comms set up

theHurc, 2015-09-23:
This following code works and I can correctly call incomingServiceRequestQueue.callOne() later in the code to service just one callback.
```
ros::CallbackQueue incomingServiceRequestQueue ;
n.setCallbackQueue(&incomingServiceRequestQueue);

ros::ServiceServer service = n.advertiseService("add_two_ints", callbackFunction);
```
Switching the order of the blocks like this doesn't set the new queue.
```
ros::ServiceServer service = n.advertiseService("add_two_ints", callbackFunction);

ros::CallbackQueue incomingServiceRequestQueue ;
n.setCallbackQueue(&incomingServiceRequestQueue);
```
I was able to determine when using the code in the second block, spinOnce calls all of the callback functions meaning the new queue was not set properly.

dirk-thomas, 2015-09-24:
The currently set callback queue will be added to the `AdvertiseOptions` of your service when it is registered but does not have a custom one set already. Consequently `setCallbackQueue` will only affect future entities but not change previously registered ones.

If you think this should be mentioned specifically in the API docs of the function a PR with this added information would be appreciated.

theHurc, 2015-09-24:
Yes I believe the documentation should be updated. I will do that. Thanks.

dirk-thomas, 2016-03-11:
Have you had a chance to update the doc? If yes, can you please point to them and close this ticket.