ros/ros_comm #678: [rosbag] synchronize topics' time stamp to /clock with rosbag play

wkentaro, 2015-10-09:
I'd like to use bagfile with gazebo, the bagfile contains sensor data like pointcloud. rosbag play with --clock option and use_sim_time param does work in most cases, but does not work with gazebo, because gazebo itself publishes /clock. 

In this case, we need to synchronize the time stamps of sensor data to the clock. Currently, I create a python script which subscribes the topics from rosbag and publishes them after synchronizing to clock like https://gist.github.com/wkentaro/2b38d8eb914c729197c6

I think this kind of thing can be done by rosbag play with an option. 

Any idea about this?

dirk-thomas, 2015-10-09:
Since this ticket is more a questions than a bug / feature request I think asking it on answers.ros.org might be better.

wjwwood, 2015-10-09:
> I think this kind of thing can be done by rosbag play with an option.

@dirk-thomas I think it is a feature request? It's not clear whether it's a question or not.

@wkentaro I don't see how you expect to mix recorded data and simulated data. Do you take time from the data in the rosbag or from the time generated by Gazebo?

wkentaro, 2015-10-09:
Sorry for ambiguity, I created this ticket as a feature request, but I'm not sure whether there are any ways to satisfy this feature.

@wjwwood 
I use `/clock` topic published by Gazebo, and I'd like synchronize the `/clock` topic and time stamp in rosbag. (ex. in image header's stamp)

the command line log is like:
```
$ roslaunch baxter_gazebo baxter_world.launch  # this launches gazebo

$ rostopic info /clock  # published by gazebo

$ rosparam get /use_sim_time
true

$ rosbag play hoge.bag  tf:=tf_old # publishes pointcloud and image like /kinect2/qhd/image_color, /tf_old
```
But the time stamp between `/kinect2/qhd/image_color` and `/clock` are too different.
And I have TF_OLD warning.

wkentaro, 2015-10-09:
So, I use https://gist.github.com/wkentaro/2b38d8eb914c729197c6 this script and transforms the time stamp of topics published by rosbag.

wjwwood, 2015-10-12:
> and I'd like synchronize the /clock topic and time stamp in rosbag. (ex. in image header's stamp)

That would require the rewriting of the message data, which rosbag is not setup to do. rosbag doesn't even deserialize messages. I'd recommend not using `--clock` with rosbag, writing a node which receives messages from rosbag, takes the current non-wall ros time (driven by `/clock`), modifies the image header's stamp and then republishes them.

A more realistic solution would be to have the gazebo plugin take `/clock` rather than produce it and use the `/clock` to populate out going header stamps. I think this is better because gazebo is generating data which it knows about and can affect, where as rosbag has no knowledge of the messages it is publishing and cannot easily modify them.

wkentaro, 2015-10-12:
So the solutions are belows, aren't they? I think the second one you said is same as modifying `/clock` to start not from 0.

* another tool which modifies these topics header
* another gazebo plugin which starts `/clock` from a specified clock time not 0. (gazebo publishes `/clock` from 0 sec)

Personally, I think the former is simpler and can be used for multiple bag files.
Where can I send PR about the script if I create it?

wjwwood, 2015-10-12:
I disagree that the former is simpler, but you can pursue which ever you like. `rosbag` does not have the ability to deserialize the messages it is publishing. Therefore it must happen in a separate tool and therefore should probably be in a separate package and not proposed for merging into `ros_comm`.

The modification of the gazebo plugin I suggested would mean that gazebo is not publishing a clock, but rather that it is depending on another source for the `/clock`, like rosbag. This approach would also work with multiple rosbags.

wkentaro, 2015-10-12:
> The modification of the gazebo plugin I suggested would mean that gazebo is not publishing a clock, but rather that it is depending on another source for the /clock, like rosbag. This approach would also work with multiple rosbags.

I understood what you mean, and now I think the latter one is simpler. I will try it.

closed 2025-05-31 by sloretz