ros/ros_comm #681: unregister() method to message_filter.Subscriber on python

wkentaro, 2015-10-12:
Do you have plan to implement this?
Or there is other way to achieve this?

dirk-thomas, 2015-10-12:
There is no plan for feature development in this repository (see the status in the rosdistro: https://github.com/ros/rosdistro/blob/00524f6409253449fdba6ee595236cd85f476ab3/jade/distribution.yaml#L3128).

Since the `rospy.Subscriber` is accessible as a member variable `sub` have you tried to call its `unregister()` method?

wkentaro, 2015-10-12:
Yeah, that works. But `message_filters.Subscriber.unregister()` is more natural, isn't it?
roscpp has that method.

wjwwood, 2015-10-12:
So it's just `message_filters.sub.unregister()` versus `message_filters.Subscriber.unregister()`? If you propose a pull request with what you think is a better API, then we'll consider it, but changes to the public API will likely be delayed until Kinetic.

wkentaro, 2015-10-12:
I'm thinking add just one method.
https://github.com/ros/ros_comm/pull/683

PR #683 (2015-10-13): [message_filters] unregister method for Subscriber
Closes #681
commit 2a00e293 (2015-10-13): [message_filters] unregister method for Subscriber

Closes #681
closed 2015-10-13 by dirk-thomas