ros/ros_comm #688: rosconsole tied to NodeHandles ?!?

rhaschke, 2015-10-20:
I was hunting down an issue for more than one hour. Having the following code snippet:
```cpp
int main(int argc, char *argv[]) {
   ros::init (argc, argv, "foo");
   rdf_loader::RDFLoader rdf("robot_description");
   ROS_ERROR("this message will never show");
}
```
As indicated, the error message after calling the RDFLoader constructor will never show up, although log4cxx is correctly setup.

The reason is actually, that rosconsole stops working after the last NodeHandle was destroyed - but it works without anyone!?! So, in the following stripped down snippet, messages 1 and 2 are shown, but 3 isn't. 
```cpp
#include <ros/ros.h>

int main(int argc, char *argv[]) {
	ros::init (argc, argv, "foo");
	ROS_ERROR("1");
	{
		ros::NodeHandle nh("~");
		ROS_ERROR("2");
	}

	ROS_ERROR("3");
	return EXIT_SUCCESS;
}
```
This behaviour isn't at all expected. Why should messaging stop working when the last NodeHandle is destroyed? Messaging might be required afterwards too. From my very first example, its really really hard to conclude, that there should be a NodeHandle beforehand. 
If this isn't considered a bug, an appropriate WARNING should be placed on the [wiki](http://wiki.ros.org/rosconsole).

wjwwood, 2015-10-21:
This might be a bug, but I guess you're talking about the messages showing up in the console and not on the `/rosout` topic?

I'm not sure if the network logging will work unless a valid node exists. The part that sounds like a bug to me is that 1 shows but 3 doesn't. I haven't looked into the implementation details to understand why that might be the case.

rhaschke, 2015-10-21:
Indeed on `/rosout` only 2 is published. However, I would expect the local console output to work in any case.

dirk-thomas, 2015-12-10:
Another example with a very unexpected outcome:

```
ros::init(argc, argv, "test");

ROS_INFO("1");
{
  ros::NodeHandle nh("foo");
  ROS_INFO("2");
}
ROS_INFO("3");
{
  ros::NodeHandle nh("bar");
  ROS_INFO("4");
}
ROS_INFO("5");
```

Only `1` and `2` will be printed.

--

I will describe why this all happens but I don't think this is fixable:

When the last `NodeHandle` instance goes out of scope a global [shutdown](https://github.com/ros/ros_comm/blob/d5644f2fcc49d29ed35f8dae8a88522f61000027/clients/roscpp/src/libros/init.cpp#L573-L607) function is being triggered in ROS. IN this process all logging infrastructure is being shut down.

Due to various design limitation it is currently not possible to call `init()` again after `shutdown()` has been called. So even when a new NodeHandle is created afterwards the system won't behave the way it is supposed to.

If anyone would like to propose an addition to the documentation I am more then happy to provide feedback. But I fear this will need to be closed as "wontfix".

de-vri-es, 2016-07-22:
One partial "fix" might be to leave log4cxx configured to print to `stdout` / `stderr` and log files atleast. The only thing that needs to be unconfigured is publishing the log messages on a ROS topic, right?

It does not solve the underlying issue, but it does improve behaviour.

dirk-thomas, 2016-07-22:
The `shutdown` function performs some cleanup and deallocation which is important (e.g. for log4cxx https://github.com/ros/ros_comm/blob/kinetic-devel/tools/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp#L360-L371). But your workaround could do the following: when no log4cxx appender is configured just print the log message to the console. While that would have different behavior then before (no more utilizing the log4cxx configuration) I agree that it might help users to notice. It would be great if someone would work on a pull request for this improvement.

de-vri-es, 2016-07-22:
There might be another option to actually address the underlying issue: use the destructor of an object with static storage duration for calling `ros::shutdown()` instead of the destruction of the last node handle. That would delay shutdown until the program ends (after `main()` returns).

I'm not generally a fan of doing things like this because it introduces global state, but there is global state as it is already.

It could look like this, in any source file linked into the core roscpp library:
```
#include <ros/ros.h>
namespace {
  class ImplCleanup {
    ~ImplCleanup() {
      ros::shutdown()
    }
  } _impl_cleanup;
}
```

One consequence of this is that it is impossible to use any ROS functionality in the destructor of other static storage objects, because the order of static initialization (and thus destruction) is not guaranteed and shutdown may already have happened when another static object's destructor is called. This seems like a very strange edge case though that *probably* doesn't occur in reality (certainly nobody *should* be doing things like that). Still, it is something to keep in mind.

I can experiment with this, if desired. See if it works as expected under some situations (thinking nodelets here). Alternatively, if it wouldn't be merged anyway I won't waste my time on it.

dirk-thomas, 2016-07-25:
A pull request would definitely be appreciated. I would expect it needs to support the following cases:

* continue to print log messages to the console and / or file after the last node has been shutdown (`rosout` won't be available for publishing anymore)
* cleanup all resources before the program exits (but after `main()` returns instead of on destruction of last node handle)
* still works the same way if a new node handle is created after the last one is destroyed

de-vri-es, 2016-07-25:
> still works the same way if a new node handle is created after the last one is destroyed

Do you mean that the current behaviour should be maintained? If I understand correctly, with current behaviour the new node handle wont be valid/usable. However, if cleanup is delayed until after main returns, it would be possible to support making a working node handle regardless of previous destroyed node handles (until main returns).

dirk-thomas, 2016-07-25:
I meant that the logging should work for a newly created node handle. So if rosout logging needs to be disabled when the first node handle is destroyed it should be reenabled when the new node handle is available.

de-vri-es, 2016-07-25:
Ah I see. I must admit I haven't looked into the details yet. But I will keep it in mind :)

cpagravel, 2017-04-13:
+1 on this bug. In my rostest SetUp() function, I create an object which has a ros::NodeHandle. After the first test finishes and that object gets destructed, I don't see console output anymore.

cpagravel, 2017-04-13:
I should add that this was causing my rostests to occasionally fail with a `std::bad_alloc` exception (very often on 16.04, less so on 14.04) when there was more than one `TEST_F()` for a test that has a node handle created somewhere in the fixture constructor and where that nodehandle is the only one.

de-vri-es, 2017-04-22:
PR #1035 should address the problems mentioned here.

It doesn't make sure log statements continue to be printed if `ros::shutdown()` is called, but it does delay automatic `ros::shutdown()` until static clean-up. A fallback for log messages may still be useful, but that can be a separate PR.

mortenfyhn, 2021-06-08:
I'm still seeing this issue. Running Melodic. Reproduced by similar code:

```cpp
#include <ros/ros.h>

int main(int argc, char** argv)
{
    ros::init(argc, argv, "node");

    {
        auto nh = ros::NodeHandle{};  // Immediately goes out of scope.
    }

    ROS_WARN("oh no");  // This does not show up in the console.

    ros::spin();
}
```

with launch file

```xml
<launch>
    <node pkg="sandbox" name="node" type="node" output="screen"/>
</launch>
```

* If I don't create a nodehandle at all, the print works.
* It doesn't seem to matter whether I block with `ros::spin()` or just let the node terminate.

de-vri-es, 2021-06-08:
Yeah, nothing was changed because of backwards compatibility issues. You just have to keep a `NodeHandle` alive. Any `NodeHandle` will do. See the conversation at https://github.com/ros/ros_comm/pull/1035 for more info.

mortenfyhn, 2021-06-08:
Got it, thanks. Maybe this should be closed as a wontfix? Or whatever is most clear for the next person stumbling upon this.