ros/ros_comm #693: roslaunch dependency on rospkg >=1.0.37 is not encoded or handled properly

trainman419, 2015-11-03:
I'm working on some modifications to roslaunch; I started with the most recent version of roslaunch from the indigo-devel branch, but I'm getting errors when I try to run it:

```
Scanning dependencies of target run_tests_aw_planner_tests_rostest_tests_circle.launch
... logging to /home/default/.ros/log/rostest-zoox15232-1193.log
Traceback (most recent call last):
  File "/home/default/zoox_ros_ws/src/driving/external/rostest/scripts/rostest", line 36, in <module>
    rostestmain()
  File "<string>", line 268, in rostestmain
  File "/home/default/zoox_ros_ws/src/driving/external/rostest/src/rostest/rostest_main.py", line 154, in rostestmain
    testCase = rostest.runner.createUnitTest(pkg, test_file, options.local, options.clear)
  File "/home/default/zoox_ros_ws/src/driving/external/rostest/src/rostest/runner.py", line 225, in createUnitTest
    config = roslaunch.parent.load_config_default([test_file], None)
  File "/home/default/zoox_ros_ws/src/driving/external/roslaunch/src/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/home/default/zoox_ros_ws/src/driving/external/roslaunch/src/roslaunch/xmlloader.py", line 730, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/home/default/zoox_ros_ws/src/driving/external/roslaunch/src/roslaunch/xmlloader.py", line 702, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/home/default/zoox_ros_ws/src/driving/external/roslaunch/src/roslaunch/xmlloader.py", line 666, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/home/default/zoox_ros_ws/src/driving/external/roslaunch/src/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/home/default/zoox_ros_ws/src/driving/external/roslaunch/src/roslaunch/xmlloader.py", line 587, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/home/default/zoox_ros_ws/src/driving/external/roslaunch/src/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/home/default/zoox_ros_ws/src/driving/external/roslaunch/src/roslaunch/substitution_args.py", line 353, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/home/default/zoox_ros_ws/src/driving/external/roslaunch/src/roslaunch/substitution_args.py", line 368, in _resolve_args
    resolved = commands[command](resolved, arg, args, context)
  File "/home/default/zoox_ros_ws/src/driving/external/roslaunch/src/roslaunch/substitution_args.py", line 144, in _find
    source_path_to_packages = rp.get_custom_cache('source_path_to_packages', {})
```

It looks like `get_custom_cache` was introduced in rosdep quite recently.

I've run `rosdep install --from-paths src -i -y -r` on my workspace, but it does not install the correct version of python-rosdep.

wjwwood, 2015-11-03:
`rosdep install ...` doesn't consider the versions of packages installed, i.e. it ignores any version constraints in `package.xml` `<*depend>` tags.

trainman419, 2015-11-03:
For reference, my machine currently has python-rospkg version 1.0.35-1 installed.

wjwwood, 2015-11-03:
I think this is the corresponding `rosdep` issue (for the version enforcing): https://github.com/ros-infrastructure/rosdep/issues/325

trainman419, 2015-11-03:
There is no corresponding `version_gte` tag in the roslaunch manifest either.

wjwwood, 2015-11-03:
Obviously that doesn't address the core issue which is the missing version dependency, but I just wanted to make sure you were aware that `rosdep` would update the package even if it were.

dirk-thomas, 2015-11-03:
Good catch @trainman419.

Because of https://github.com/ros/ros_comm/commit/f2a5838b1243a8859d66ac19fa8f0df3751509dd#diff-85d266eabfa0eb4dda9a4551c1929065R137 roslaunch needs to declare a minimum version of ropskg.

commit 4ee44bdc (2015-11-03): add missing minimum version for rospkg dependency (fix #693)
closed 2015-11-03 by dirk-thomas