ros/ros_comm #695: Non-Ascii Characters in msg File Comments Cause Python Runtime Error

evenator, 2015-11-06:
If a .msg file contains non-ascii characters, catkin will still generate a ROS message from the file with no warnings or errors. roscpp nodes can publish the message type with no warnings or errors. However, attempting to echo the message with rostopic results in rostopic emitting a Python stacktrace and dying.

The exception is SyntaxError: Non-ASCII character <ascii character> in file <lib_path>/python2.7/dist-packages/non_ascii_msgs/msg/_<MsgName>.py on line <line>, but no encoding declared; see http://www.python.org/peps/pep-0263.html for details

I've tested this on both Hydro and Indigo. See my minimal reproduction at https://github.com/evenator/rosmsg-Non-ASCII

dirk-thomas, 2015-11-06:
First the example package needs the following line in CMake in order to build correctly in its own workspace:

    include_directories(${catkin_INCLUDE_DIRS})

But I can't reproduce the problem. `rostopic echo /test` works fine for me on Indigo.

Can you please check what `locale` you have set on your machine?

evenator, 2015-11-06:
I fixed the CMake issue (which, oddly, doesn't seem to present when building with `catkin_tools`).

Here's `locale` from a machine I tested with. (ROS Indigo on Ubuntu 14.04 desktop, python version 2.7.6.)

    $ locale
    LANG=en_US.UTF-8
    LANGUAGE=en_US
    LC_CTYPE="en_US.UTF-8"
    LC_NUMERIC="en_US.UTF-8"
    LC_TIME="en_US.UTF-8"
    LC_COLLATE="en_US.UTF-8"
    LC_MONETARY="en_US.UTF-8"
    LC_MESSAGES="en_US.UTF-8"
    LC_PAPER="en_US.UTF-8"
    LC_NAME="en_US.UTF-8"
    LC_ADDRESS="en_US.UTF-8"
    LC_TELEPHONE="en_US.UTF-8"
    LC_MEASUREMENT="en_US.UTF-8"
    LC_IDENTIFICATION="en_US.UTF-8"
    LC_ALL=

dirk-thomas, 2015-11-06:
You might want to fill a ticket against `catkin_tools` regarding the include directories.

Can you confirm that it works for you when you add the following line at the very top of the generated file `_Test.py`?

    # This Python file uses the following encoding: utf-8

evenator, 2015-11-07:
Confirmed. Adding that line to the beginning of the file allows me to echo the topic with rostopic echo.

Tested with ROS Indigo on Ubuntu 14.04 with Python 2.7.6.

closed 2015-11-09 by dirk-thomas

dirk-thomas, 2015-11-10:
While this immediate problem has been fixed I would recommend you not to use non-ascii characters in message / service definitions. The likeliness of problems in other existing code is pretty high.

evenator, 2015-11-10:
Yeah, I certainly wasn't planning on it. I copied a docstring out of a PDF for a comment and happened upon the issue.