ros/ros_comm #699: launch-prefix removes quotation on roslaunch

furushchev, 2015-11-10:
When launching nodes with `launch-prefix`, I have a problem.
I want to test a node with X window on server without display.
I created launch file like below:

```xml
<launch>
  <test test-name="eusroseus" pkg="roseus" type="roseus"
        launch-prefix="xvfb-run --server-args='-screen 0, 1024x768x24' " />
</launch>
```

but the actual command passed to shell was

```bash
xvfb-run --server-args=-screen 0, 1024x768x24 /workspace/ros/ws_jsk_roseus/src/jsk_roseus/roseus/bin/roseus /workspace/ros/ws_jsk_roseus/src/jsk_roseus/roseus/test/test-roseus.l --gtest_output=xml:/workspace/ros/ws_jsk_roseus/build/roseus/test_results/roseus/rosunit-eusroseus.xml __name:=eusroseus __log:=/workspace/jsk-ros-pkg/jsk_roseus/log/61bcf9a8-87b7-11e5-88b7-0242ac110018/eusroseus-2.log 
```

The problem is roslaunch cuts `'` of `--server-args='`.

I look deeper the code and found that `shlex.parse` removes all quotation of `launch-prefix`:

```python
import shlex
shlex.split("xvfb-run --server-args='-screen 0, 1024x768x24'")
> ['xvfb-run', '--server-args=-screen 0, 1024x768x24']
```

Is this a correct behavior or bug?
If this is correct, is there any way to pass a quote on `launch-prefix`?

dirk-thomas, 2015-11-10:
As far as I can see `shlex.split` does the right thing in your example. It keeps the `server-args` together and doesn't split them on the spaces inside the argument. (Otherwise you would get: `['xvfb-run', '--server-args=-screen', '0,', '1024x768x24']`)

But if the result is later passed to an invocation which again perform something like `shell=True` then the individual parts will we interpreted as separate arguments. So the problem is likely where this result is being used rather then the `shlex` function itself.

furushchev, 2015-11-11:
@dirk-thomas Thank you for answer.

I forgot to attach where `shlex.split` is used on roslaunch
https://github.com/ros/ros_comm/blob/ec86fc0790643ce7737686d1375b9ba085db7678/tools/roslaunch/src/roslaunch/node_args.py#L201

I managed to avoid this problem by using short option like below:

```
<launch>
  <test test-name="eusroseus" pkg="roseus" type="roseus"
        launch-prefix="xvfb-run -s '-screen 0, 1024x768x24' " />
</launch>
```

but it might occurs the problem in the future.

closed 2015-11-11 by furushchev