ros/ros_comm #703: ABI compatibility broken between 1.11.13 and 1.11.16 versions (for indigo)

jonbinney, 2015-11-20:
After updating moveit from debs today (but not updating roscpp) I got the following error:
```
/opt/ros/indigo/lib/libmoveit_planning_scene_monitor.so: undefined reference to `ros::WallTimer::setPeriod(ros::WallDuration const&, bool)'                         
collect2: error: ld returned 1 exit status   
```

Looks like this is a result of broken ABI compatilibty between 1.11.13 and 1.11.16, which caught me off guard since the version number didn't change much.

Output of `git diff 1.11.13 1.11.16 -- clients/roscpp/include/ros/timer.h`:

```diff
diff --git a/clients/roscpp/include/ros/timer.h b/clients/roscpp/include/ros/timer.h
index ec3f1c3..6849875 100644
--- a/clients/roscpp/include/ros/timer.h
+++ b/clients/roscpp/include/ros/timer.h
@@ -67,8 +67,9 @@ public:
 
   /**
    * \brief Set the period of this timer
+   * \param reset Whether to reset the timer. If true, timer ignores elapsed time and next cb occurs at now()+period
    */
-  void setPeriod(const Duration& period);
+  void setPeriod(const Duration& period, bool reset=true);
 
   bool isValid() { return impl_ && impl_->isValid(); }
   operator void*() { return isValid() ? (void*)1 : (void*)0; }
@@ -99,7 +100,7 @@ private:
 
     bool isValid();
     bool hasPending();
-    void setPeriod(const Duration& period);
+    void setPeriod(const Duration& period, bool reset=true);
 
     void start();
     void stop();
```

@dirk-thomas you can probably just close this issue right away since I don't see anything to be done, but I wanted to create this issue in case others ran into problems after updating some packages after today's sync to public (it took me some time to track this down).

/cc @wjwwood

jonbinney, 2015-11-20:
(to anyone who sees that linker error, just do a `sudo apt-get dist-upgrade` to make sure you get new versions of all packages that depend on roscpp)

closed 2015-11-20 by jonbinney

dirk-thomas, 2015-11-20:
Just for the record: this will be the same with Jade.