ros/ros_comm #706: Latched Topics in Split Bags

DLu, 2015-11-25:
I'm trying to record TF data across split bags. (`rosbag record --split --duration=5s /tf /tf_static`) The /tf_static topic is only recorded in the first bag. 

Is there anyway to force rosbag to record latched topics in each bag?

dirk-thomas, 2015-11-25:
There is currently no such option.

wjwwood, 2016-05-17:
I outlined some possible solutions in https://github.com/ros/ros_comm/issues/805#issue-155150413

JanezCim, 2021-02-18:
Just an FYI for anyone struggling, using the command

`rosbag play 0.bag 1.bag 2.bag`

will keep all the static transforms correctly for the whole duration of all the split bag files combined. The problem becomes when you try to play them back separately. But afaik this is being solved in Noetic.

chaoflow, 2021-10-06:
This has been implemented with #1850 included in noetic. I think the issue can be closed.

closed 2021-10-06 by DLu