ros/ros_comm #709: reenable rosmsg tests

dirk-thomas, 2015-12-01:
Current the tests in the `rosmsg` are not being used: https://github.com/ros/ros_comm/tree/indigo-devel/tools/rosmsg/test

gerkey, 2015-12-10:
Mostly done in ea7cf134b8c429ffacf8cac84e374207ec8e5220.

There are a few remaining failures, which might be actual problems. I'll keep looking into it.

gerkey, 2015-12-10:
FYI, here are the remaining failures:

~~~
======================================================================
FAIL: test_get_msg_text (test_rosmsg.TestRosmsg)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/gerkey/code/rosmsg-tests/src/ros_comm/tools/rosmsg/test/test_rosmsg.py", line 74, in test_get_msg_text
    self.assertEquals(text, rosmsg.get_msg_text(type_, raw=True))
AssertionError: 'test_rosmaster/Empty empty\n' != 'Empty empty\n'

======================================================================
FAIL: test_get_srv_text (test_rosmsg.TestRosmsg)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/gerkey/code/rosmsg-tests/src/ros_comm/tools/rosmsg/test/test_rosmsg.py", line 136, in test_get_srv_text
    self.assertEquals(text, rosmsg.get_srv_text('test_rosmaster/RossrvB', raw=True))
AssertionError: 'test_rosmaster/Empty empty\n---\ntest_rosmaster/Empty empty\n' != 'Empty empty\n---\nEmpty empty\n'

======================================================================
FAIL: test_cmd_show (test_rosmsg_command_line.TestRosmsg)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/gerkey/code/rosmsg-tests/src/ros_comm/tools/rosmsg/test/test_rosmsg_command_line.py", line 156, in test_cmd_show
    self.assertEquals(text, output)
AssertionError: 'Empty empty\n\n' != 'test_rosmaster/Empty empty\n\n'

----------------------------------------------------------------------
~~~

dhood, 2016-01-29:
I've been looking into this to see if I could help - 

have opened https://github.com/ros/ros_comm/pull/734 with one way of fixing remaining failures

dirk-thomas, 2016-03-11:
Addressed by PR #751. Thank you.

closed 2016-03-11 by dirk-thomas