ros/ros_comm #725: roslaunch --ros-args Fails on Implicit Arguments

evenator, 2016-01-08:
The following simple launch file has 3 arguments:
```xml
<launch>
    <arg name="optional" doc="An optional argument" default="1.0" />
    <arg name="explicit_required" doc="A required argument" />
    <node name="$(arg implicit_required)" pkg="foo" type="bar" />
</launch>
```
The `optional` and `explicit_required` arguments are specified explicitly, and `implicit_required` is specified implicitly.
Running `roslaunch --ros-args` (ROS Indigo) on this launch file with no arguments supplied results in an error:
```
$ roslaunch --ros-args /tmp/test_args.launch
[/tmp/test_args.launch] requires the 'implicit_required' arg to be set
```
Setting the argument like so, results in the following usage stanza, which doesn't include the implicit argument:
````
$ roslaunch --ros-args /tmp/test_args.launch implicit_required:=val
Required Arguments:
  explicit_required: A required argument
Optional Arguments:
  optional (default "1.0"): An optional argument
````

dirk-thomas, 2016-01-08:
According to http://wiki.ros.org/roslaunch/XML#substitution_args `$(arg foo)` evaluates to the value specified by an `<arg>` tag. Why do you assume that you can implicitly specify arguments like in your example. Afaik you must specify the arg using an `<arg>` tag. So your example launch file has only *two* arguments.

evenator, 2016-01-08:
> Why do you assume that you can implicitly specify arguments like in your example. Afaik you must specify the arg using an <arg> tag. So your example launch file has only two arguments.

`roslaunch` allows implicit arguments (see below). While I agree that implicit arguments are bad practice (I never use them personally), I would assume that this flag would document them.
If implicit arguments are not valid roslaunch, the tool should not allow them. If implicit arguments are not allowed, close this and I can file a separate issue for that.

```xml
<launch>
    <arg name="optional" doc="An optional argument" default="1.0" />
    <arg name="explicit_required" doc="A required argument" />
    <param name="$(arg implicit_required)" value="foo" />
</launch>
```

```
$ roslaunch /tmp/test_args.launch implicit_required:=test
... logging to /home/evenator/.ros/log/c3440c1a-b625-11e5-ab7a-6431502365b0/roslaunch-boomer-11964.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/evenator/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://boomer.dyn.datasys.swri.edu:34245/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /test: foo

NODES

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
```

dirk-thomas, 2016-01-08:
Obviously implicit arguments work in this example. But I don't think it is feasible to expect them to be extracted automatically.

Currently that would require that the whole xml file is considered to find `$(arg ...)` expressions. But especially if you consider that future enhancement will likely allow nested expressions (e.g. `$(arg $(something else))`) the complexity becomes enormous. At that point it would be necessary to fully evaluate expression to determine if there are implicit arguments.

Therefore I don't think either of the two options is feasible:
* improve roslaunch to also show implicit arguments
* prevent implicit arguments (since existing code will rely on the existing behavior)

evenator, 2016-01-08:
> prevent implicit arguments (since existing code will rely on the existing behavior)

If you're not comfortable with disabling this "feature," perhaps roslaunch should emit a warning message or something to discourage its use or indicate that it's deprecated?