ros/ros_comm #726: roslaunch --ros-args fails too early for required arguments

jatowler, 2016-01-08:
## Summary

The `roslaunch --ros-args [package] <filename>` invocation should provide information about the arguments defined in a launch file. However, in the very situation which in which this flag would be most helpful--determining what arguments are required to use the file--it instead fails, complaining that a required argument is not set.

This issue is distinct from #231; see "Related Issues" below.

## Example

Under ROS Indigo, using the below `test.launch` file, `roslaunch --ros-args test.launch` reports the following:

```
± cat test.launch 
<?xml version="1.0"?>
<launch>
  <arg name="required" doc="This argument is required"/>
  <param name="arg_user" value="$(arg required)"/>
</launch>

± roslaunch --ros-args test.launch
[test.launch] requires the 'required' arg to be set
```

Obviously, this can be remedied by providing the argument, like so:

```
± roslaunch --ros-args test.launch required:="foo"
Required Arguments:
  required: This argument is required
```

But consider a real system with a launch file with many arguments. In this case, instead of documenting the required and optional arguments so the launch file can be used, perhaps as an inclusion in another file, `roslaunch` provides exactly one missing argument on every invocation, requiring the user to slog through repeated (and different) invocations to determine the full extent of the launch file's required arguments. This behavior negates any practical utility of `--ros-args` when a launch file has required arguments, as it provides literally no more information than a bare `roslaunch` invocation.

## Expected Behavior

Using the above example, the `--ros-args` flag should ignore failures that (rightly) abort normal `roslaunch` invocations and report the arguments regardless of whether they have been set.

## Likely Causes

`roslaunch --ros-args` follows the same code path as every other `roslaunch` invocation and reports an error when a required argument is not set. However, that is inappropriate in this case, as discussed above.

## Related Issues

This issue appears to be related to #231, but it is distinct. In that case, another `arg` tag uses the required argument and `roslaunch` fails during argument parsing. In this case, a non-`arg` tag uses the required argument (in the example, a `param` tag, but it could be any tag), and a different failure occurs. Note that the ideal resolution is the same as #231, in that `roslaunch` should not fail when it encounters an undefined argument when `--ros-args` is specified. Probably the same difficulties with the fix arise as in that issue, but, again, this issue is distinct, so it is reported separately.

Note also that #379 does not fix this issue (actually, it also appears it does not fix #231, as changing the `param` tag in `test.launch` to an `arg` tag results in the same error reported there).

wkentaro, 2016-04-26:
Same problem here, I use `roslaunch-check` but it fails because of this problem.

dirk-thomas, 2016-04-26:
Please consider to create a pull request to use dummy arguments for `$(arg ...)`. One potential problem might be that the actual result of the arg evaluation might affect the launch file in a way. So if you use a mock value instead the result might be different.

closed 2025-05-31 by sloretz