ros/ros_comm #729: Non-virtual destructors in ros::console

trainman419, 2016-01-18:
When compiling our codebase against ros::console with the warnings turned all the way up, we get these warnings:

```
In file included from /opt/ros/indigo/include/ros/ros.h:40:0,
                 from 
                 ... < removed private file names > ...
/opt/ros/indigo/include/ros/console.h:99:7: warning: ‘class ros::console::LogAppender’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 class LogAppender
       ^
/opt/ros/indigo/include/ros/console.h:109:8: warning: ‘struct ros::console::Token’ has virtual functions and accessible non-virtual destructor [-Wnon-virtual-dtor]
 struct Token
```

wjwwood, 2016-01-18:
The destructor should be `virtual`, marked here:

https://github.com/ros/ros_comm/blob/indigo-devel/tools/rosconsole/include/ros/console.h#L99-L105

dirk-thomas, 2016-03-17:
Since this is an ABI breaking change it will have to only go into the upcoming kinetic-devel branch.

dirk-thomas, 2016-03-17:
Please see #773 for the proposed change in Kinetic.

closed 2016-03-18 by dirk-thomas