ros/ros_comm #736: rosbag filter adds two bytes per topic

chaoflow, 2016-02-03:
The bug occurs at least with ROS indigo, we did not test other versions.
cc @durko @dejanpan
```
$ rosbag info original.bag 
path:        original.bag
version:     2.0
duration:    11.0s
start:       Mar 06 2015 14:32:21.02 (1425648741.02)
end:         Mar 06 2015 14:32:32.04 (1425648752.04)
size:        60.8 KB
messages:    308
compression: none [1/1 chunks]
types:       rosgraph_msgs/Log [acffd30cd6b6de30f120938c17c593fb]
             std_msgs/String   [992ce8a1687cec8c8bd883ec73ca41d1]
topics:      /chatter       98 msgs    : std_msgs/String  
             /rosout       106 msgs    : rosgraph_msgs/Log (2 connections)
             /rosout_agg   104 msgs    : rosgraph_msgs/Log
$ rosbag filter original.bag x.bag True                  
$ rosbag filter x.bag y.bag True
$ rosbag filter y.bag z.bag True
$ du -bs *
62209	original.bag
59370	x.bag
59376	y.bag
59382	z.bag
```
The change of filesize from original.bag to x.bag is to be expected. After that the filesize (and md5sum) should remain the same. However, `rosbag filter` adds 2 bytes per topic.

At least one byte per topic we found: it's a `\n` (hex `0a`) appended to pytype._full_text:
```
$ xxd x.bag  |grep -A1 ^00014b0
00014b0: 7374 7269 6e67 2066 7261 6d65 5f69 640a  string frame_id.
00014c0: 0a27 0000 006d 6435 7375 6d3d 6163 6666  .'...md5sum=acff

$ xxd y.bag  |grep -A1 ^00014b0
00014b0: 7374 7269 6e67 2066 7261 6d65 5f69 640a  string frame_id.
00014c0: 0a0a 2700 0000 6d64 3573 756d 3d61 6366  ..'...md5sum=acf

$ xxd z.bag  |grep -A1 ^00014b0
00014b0: 7374 7269 6e67 2066 7261 6d65 5f69 640a  string frame_id.
00014c0: 0a0a 0a27 0000 006d 6435 7375 6d3d 6163  ...'...md5sum=ac
```

chaoflow, 2016-02-03:
Link to tarball with bags: https://ternaris.com/rosbag-filter-bags-issue736.tar.gz

dirk-thomas, 2016-03-28:
This has been addressed by ros/genpy#47.

closed 2016-03-28 by dirk-thomas