ros/ros_comm #742: segfault when reading rosbag

mcamurri, 2016-02-19:
Hi all,
I'm trying to parse a rosbag file the same way as it is explained [here](http://answers.ros.org/question/49489/rosbag-api-c/) but when I try to instantiate the embedded message I get a segfault. The code I'm using is as follows:
```
rosbag::Bag bag;
std::vector<std::string> topics;
std::cout << "Opening " << filename_ << std::endl;
bag.open(filename_.c_str(), rosbag::bagmode::Read);
std::cout << "Mode: " << bag.getMode() << std::endl;

topics.push_back(std::string("/Imu"));
topics.push_back(std::string("/joint_states"));
rosbag::View view(bag, rosbag::TopicQuery(topics));
foreach(rosbag::MessageInstance const m, view) {
    std::cout << "Found message of type "  << m.getDataType() << std::endl;
    if(m.getDataType().compare("sensor_msgs/Imu") == 0) {
        sensor_msgs::ImuConstPtr s = m.instantiate<sensor_msgs::Imu>();

        cout << "S = " << s << std::endl;
        double ax = s->linear_acceleration.x;
}
```

See also  related question on ros answers [here](http://answers.ros.org/question/227047/segfault-when-reading-rosbag/)

dirk-thomas, 2016-02-19:
Please provide a minimal reproducible example. In your case e.g. the bag file you are trying to rea as well as s ready to run code snippet. Without that information nobody can try what you are experiencing without significant extra work. That is probably also the reason why you didn't get any comments on the question on answers.ros.org.

mcamurri, 2016-02-25:
Dear @dirk-thomas, thanks for the advice, I will try to attach a complete example in the future.

For this issue I eventually solved myself  by de-referentiating the pointer and then using the local variable to access the data:
```
sensor_msgs::ImuConstPtr s = m.instantiate<sensor_msgs::Imu>();
sensor_msgs::Imu imu = (*s);
```

dirk-thomas, 2016-02-25:
Great, please close this issue if it is resolved.

closed 2016-02-25 by mcamurri